mantidproject/mantid #28650: Build Mantid with Qt5 only

AndreiSavici, 2020-05-05:
This is an umbrella ticket for building on systems where there is no Qt4 (Ubuntu 20.04)

* [ ] Build Mantid
* [ ] Build tests
* [ ] UnitTests pass
* [ ] Systemtests pass
* [ ] Build documentation #28647
* [ ]  Doc tests

AndreiSavici, 2020-05-05:
With changes in #28651, most of the failing tests are because we force Qt4 testing even if it's not available.

martyngigg, 2020-11-20:
@AndreiSavici The system test failures on 20.04, with the exception of PVPython, are not all required to be fixed for us to disable MantidPlot on the other platforms.
Would you be happy if we sorted out disabling MantidPlot first? I'm just aware that Release 6.0 is coming and the code freeze is the start of January so getting this is donesooner is probably better.

AndreiSavici, 2020-11-20:
@martyngigg Fine with me. The ORNL team is looking into those tests anyway, but it can be done after removing MantidPlot

martyngigg, 2020-11-20:
Brilliant. Thanks for the update.

AndreiSavici, 2020-11-20:
@martyngigg When the PR with removing MantidPlot is merged, we need to send an email to the user community to say that mantidplotnightly is gone.

martyngigg, 2021-01-07:
I've moved the remaining parts of this to a new issue that deals with actually removing MantidPlot, #30268, so I'll close this now.

closed 2021-01-07 by martyngigg