mantidproject/mantid #28652: New SANS Interface in MantidPlot 5.0.0 not loading old (2013) data

smk78, 2020-05-06:
MantidPlot 5.0.0 on W10/x64

Complete data package for testing: [For_ExPaNDS.zip](https://github.com/mantidproject/mantid/files/4586213/For_ExPaNDS.zip)

### Expected behavior
 The new SANS Interface should be intelligent enough to reduce legacy data.

### Actual behavior
Attempting to process the attached LOQ data from the batch file reduce_gfp:
[image]
results in this error:
```
Welcome to Mantid 5.0.0
Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid5.0
Downloading 1 file from the instrument repository
Instrument changed to SANS2D.
Instrument changed to LOQ.
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\batch_process_runner.py", line 98, in _load_workspaces_on_thread
    states, errors = get_states_func(row_entries=[row])
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 73, in run
    result = func(*args, **kwargs)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 922, in get_states
    states, errors = create_states(state_model_with_view_update,
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 32, in create_states
    _get_thickness_for_row(row)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 51, in _get_thickness_for_row
    file_info = row.file_information
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\RowEntries.py", line 67, in file_information
    return file_factory.create_sans_file_information(self.sample_scatter)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 1100, in create_sans_file_information
    elif is_added_histogram(full_file_name) or is_added_event(full_file_name):
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 559, in is_added_histogram
    is_added, _, is_event = get_added_nexus_information(file_name)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 522, in get_added_nexus_information
    is_added_event_file, number_of_periods_event = get_added_event_info(h5_file, top_level_keys)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 477, in get_added_event_info
    workspace_name = get_workspace_name(entry)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 453, in get_workspace_name
    return entry["workspace_name"][0].decode("utf-8")
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "C:\MantidInstall-v500\bin\lib\site-packages\h5py-2.10.0-py3.8-win-amd64.egg\h5py\_hl\group.py", line 264, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py\h5o.pyx", line 190, in h5py.h5o.open

"Unable to open object (object 'workspace_name' doesn't exist)"
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\batch_process_runner.py", line 98, in _load_workspaces_on_thread
    states, errors = get_states_func(row_entries=[row])
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 73, in run
    result = func(*args, **kwargs)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 922, in get_states
    states, errors = create_states(state_model_with_view_update,
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 32, in create_states
    _get_thickness_for_row(row)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 51, in _get_thickness_for_row
    file_info = row.file_information
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\RowEntries.py", line 67, in file_information
    return file_factory.create_sans_file_information(self.sample_scatter)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 1100, in create_sans_file_information
    elif is_added_histogram(full_file_name) or is_added_event(full_file_name):
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 559, in is_added_histogram
    is_added, _, is_event = get_added_nexus_information(file_name)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 522, in get_added_nexus_information
    is_added_event_file, number_of_periods_event = get_added_event_info(h5_file, top_level_keys)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 477, in get_added_event_info
    workspace_name = get_workspace_name(entry)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 453, in get_workspace_name
    return entry["workspace_name"][0].decode("utf-8")
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "C:\MantidInstall-v500\bin\lib\site-packages\h5py-2.10.0-py3.8-win-amd64.egg\h5py\_hl\group.py", line 264, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py\h5o.pyx", line 190, in h5py.h5o.open

"Unable to open object (object 'workspace_name' doesn't exist)"
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\batch_process_runner.py", line 98, in _load_workspaces_on_thread
    states, errors = get_states_func(row_entries=[row])
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 73, in run
    result = func(*args, **kwargs)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\presenter\run_tab_presenter.py", line 922, in get_states
    states, errors = create_states(state_model_with_view_update,
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 32, in create_states
    _get_thickness_for_row(row)
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\create_state.py", line 51, in _get_thickness_for_row
    file_info = row.file_information
  File "C:\MantidInstall-v500\scripts\SANS\sans\gui_logic\models\RowEntries.py", line 67, in file_information
    return file_factory.create_sans_file_information(self.sample_scatter)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 1100, in create_sans_file_information
    elif is_added_histogram(full_file_name) or is_added_event(full_file_name):
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 559, in is_added_histogram
    is_added, _, is_event = get_added_nexus_information(file_name)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 522, in get_added_nexus_information
    is_added_event_file, number_of_periods_event = get_added_event_info(h5_file, top_level_keys)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 477, in get_added_event_info
    workspace_name = get_workspace_name(entry)
  File "C:\MantidInstall-v500\scripts\SANS\sans\common\file_information.py", line 453, in get_workspace_name
    return entry["workspace_name"][0].decode("utf-8")
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "C:\MantidInstall-v500\bin\lib\site-packages\h5py-2.10.0-py3.8-win-amd64.egg\h5py\_hl\group.py", line 264, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py\h5o.pyx", line 190, in h5py.h5o.open

"Unable to open object (object 'workspace_name' doesn't exist)"
```
Suspecting an issue with the -add files, tried loading those through the Workspace box generic Load button. They load fine and display correctly.

Switching to the old (now deprecated) SANS Interface (still in 5.0.0), everything works fine.
[image]

gemmaguest, 2020-05-07:
@smk78 The new GUI is expecting to find a `workspace_name` entry in the nexus file. It only uses this in the check to see whether it is added event data. It's failing because this entry is not there in the old `-add` files. If you regenerate the add files it works but agree it should not fail for these old files.

I can change it to use the file name instead if it cannot find a `workspace_name` entry, or actually just handling the error more gracefully will mean the reduction works for this example. However I am not sure if this will break things for event mode. Do you have an example with added event data that I can test with?

smk78, 2020-05-07:
I can certainly dig some up for you!

smk78, 2020-05-11:
@gemmaguest I've put a complete dataset with added event data in:

ftp://ftp.nd.rl.ac.uk/scratch/Loq/For_Gemma.zip

(NB: it's is SANS2D data!)

smk78, 2020-05-11:
It did occur to me that LOQ is the only one of the ISIS SANS instruments _not_ to produce event data. However, the other instruments could, in principle (I think), produce histogram data even though they normally produce event data. So an approach to resolving this issue that does not just look at the instrument is probably the way to go.

gemmaguest, 2020-05-13:
@smk78 Thanks, this confirms that my fix doesn't break things for event data, at least for newly generated -add files. However, old event -add files from before ~2013 would still fail. However I can't see how to generate these. Going back go Mantid v2.5.3, the GUI can only save -add files as histogram workspaces, and SaveNexusProcessed crashes with workspace groups...

Anyway, I'll go ahead with my fix for histogram data because it won't make things worse, and if we do come across old event -add files we need to process I can take another look.

smk78, 2020-05-14:
> It did occur to me that LOQ is the only one of the ISIS SANS instruments _not_ to produce event data. However, the other instruments could, in principle (I think), produce histogram data even though [...]

RobD told me that the runs in LARMOR cycle_15_2 were in histogram mode. However I'm uncertain if he meant the nxs or the raw! When I tried sum the nxs it looked to me like the interface felt they were event (in respect of how it wanted to treat the binning). Can you tell?

smk78, 2020-05-14:
> @smk78 Thanks, this confirms that my fix doesn't break things for event data, at least for newly generated -add files. However, old event -add files from before ~2013 would still fail. However I [...]

I seem to recall that adding event files on SANS2D was an issue for some time and was done by a script. I'll have a rummage around!

smk78, 2020-05-14:
This looks promising @gemmaguest 

[SANSadd_standalone.zip](https://github.com/mantidproject/mantid/files/4627538/SANSadd_standalone.zip)

smk78, 2020-05-15:
> @smk78 Thanks, this confirms that my fix doesn't break things for event data, at least for newly generated -add files. However, old event -add files from before ~2013 would still fail.

Sarah has had a hunt around and says:

[The files below] _are from the first cycle we ran_ [on SANS2D] _with event mode (11/3) and they are stopped-flow samples...  they were run on 10th October 2011.  As they are stopped-flow then I assume we added them in event mode (with a time shift?)._

_I see at the end of this experiment I also ran the test AOT + NaCl samples (which I used to check that the event mode was working in terms of both the DAE and Mantid) and the added file for this also appears to be there - 10198 and it was created on 18th October 2011._

\\\isis\inst$\NDXSANS2D\User\Processed\SANS2D\SANS2D00010163-add.nxs 

\\\isis\inst$\NDXSANS2D\User\Processed\SANS2D\SANS2D00010175-add.nxs 

\\\isis\inst$\NDXSANS2D\User\Processed\SANS2D\SANS2D00010181-add.nxs 

\\\isis\inst$\NDXSANS2D\User\Processed\SANS2D\SANS2D00010198-add.nxs

closed 2020-05-15 by martyngigg

gemmaguest, 2020-05-22:
@smk78 Thanks for digging out those examples. It looks like those `-add` files even for event data have all been saved as histograms, so I don't think we will run in to any problem with event data. I'll therefore leave this issue closed. Of course, let me know if you do hit any further problems.

PR #28659 (2020-05-15): Fix a bug loading old "-add" files
**Report to:** Steve King

This PR removes an assumption that a `workspace_name` entry exists in SANS `-add` files. Old data produced from mid-2013 and earlier do not have this entry and the SANS GUI was throwing due to this assumption. The fix is to use the file name instead if the workspace_name does not exist, although it is somewhat irrelevant what is used because the check is only to see whether it is event data and I don't believe we would have generated event data `-add` files from before 2013 (see discussion on the issue).

Fixes #28652

**To test:**
 [... cut]