mantidproject/mantid #28654: Unicode WS axis labels Hard Crash

DanielMurphy22, 2020-05-06:
**Original reporter:** [James Lord](https://forum.mantidproject.org/t/unicode-in-workspace-axis-labels/615)

### Expected behavior
Since Migration to Python 3, the code below should work.
### Actual behavior
It runs but you get a very hard crash of Mantid when you "Show Data"
### Steps to reproduce the behavior
Run this code:
```
W=CreateWorkspace(DataX=[0,1,2],DataY=[3,7,5],DataE=[0.2,0.3,0.1],NSpec=1)

lbl=W.getAxis(0).setUnit("Label")
lbl.setLabel("Time",u"\u00B5s") # microseconds
W.setYUnitLabel(u"Distance in \u00C5") # Angstrom symbol
```
Now "Show Data" of the Workspace
Many Many Error reporters...!
### Platforms affected
I found this on Workbench 5.0 on Mac, but presumably effects all platforms

thomashampson, 2021-03-26:
It doesn't crash on Windows, but the micro symbol doesn't display properly in the table
[image]

thomashampson, 2021-04-06:
Although there is a pending pull request with a fix for this problem, in the meantime a small modification of the script will circumvent the issue:
```
from mantid.simpleapi import *
from mantid.kernel import UnitLabel

W=CreateWorkspace(DataX=[0,1,2],DataY=[3,7,5],DataE=[0.2,0.3,0.1],NSpec=1)

lbl=W.getAxis(0).setUnit("Label")
lbl.setLabel("Time", UnitLabel("microseconds", "\u00B5s", "\u00B5s")) # microseconds
W.setYUnitLabel(u"Distance in \u00C5") # Angstrom symbol
```

PR #31036 (2021-04-16): Unicode axis label crash
**Description of work.**
The following script was causing problems because the second argument in the `setLabel()` function call was implicitly constructing an object that assumed it was being passed an ASCII string:
```
from mantid.simpleapi import *

W=CreateWorkspace(DataX=[0,1,2],DataY=[3,7,5],DataE=[0.2,0.3,0.1],NSpec=1)
lbl=W.getAxis(0).setUnit("Label")
lbl.setLabel("Time",u"\u00B5s") # microseconds
W.setYUnitLabel(u"Distance in \u00C5") # Angstrom symbol
```
When right clicking the workspace and clicking "Show Data" the column headers were garbled, and crashing  [... cut]
closed 2021-04-16 by gvardany