mantidproject/mantid #28666: Hide confusing intermediate indirect workspaces.

Mantid-Matthew, 2020-05-12:
Currently the Indirect FQ fit tab creates a HWHM workspace whenever data is loaded to used for fitting. This creates confusion on the user end and means that the workflows are more difficult to replicate in scripts. This functionality should therefore be moved into the fitting algorithms.

Mantid-Matthew, 2020-09-01:
@Harrietbrown This issue is related to the F(Q) fitting issues you have been working on so re-assigning to you. Let me know if more info is needed.

Harrietbrown, 2021-05-24:
not stale

Harrietbrown, 2021-11-30:
Not stale

Harrietbrown, 2022-06-07:
Not stale

Harrietbrown, 2022-12-05:
Not Stale

Harrietbrown, 2022-12-06:
Not stale

Harrietbrown, 2023-01-24:
[image] [Hide _HWHM workspaces produced by FqFit https://github.com/mantidproject/mantid/issues/28666](https://trello.com/c/gSMuBoG2/9-hide-hwhm-workspaces-produced-by-fqfit-https-githubcom-mantidproject-mantid-issues-28666)

AnthonyLim23, 2023-01-30:
making it a hidden workspace will break the workspace history. It would probably be better to delete the ws at the end of the processing step.

Harrietbrown, 2023-01-30:
For this we can move the creation of the HWHM into when the run button is clicked and deleate at the end, but we will still need a a workspace that can be plotted.

AnthonyLim23, 2023-07-31:
not stale

robertapplin, 2023-12-01:
Still one last part of this needs to be completed

robertapplin, 2023-12-04:
I've considered a couple of potential solutions for this issue, with the first option being the preferred approach, while the second one, although functional, is not an ideal fix.

**Solution 1:**
To address the concern of the HWHM workspace being output into the ADS, the proposal is to store it solely in memory. However, implementing this solution requires a substantial effort to decouple the Data Analysis interface code from the ADS. The advantages include:

- Independence from the ADS singleton, leading to more better designed code and theoretically reducing bugs.
- Ensuring the "Script to Clipboard" feature for result workspaces will reliably work.

On the downside:

- Significant effort is needed to disentangle the Data Analysis interface from the ADS.
- The QENSFitSequential and QENSFitSimultaneous "Input" property would require re-work for this approach.
- Users will lose the ability to view intermediate workspaces during their workflow for verification.

**Solution 2:**
This alternative involves using a hidden workspace for the HWHM workspace in the ADS by prefixing "__" to its name. While it achieves the goal of concealing the HWHM workspace and is a quick implementation, it comes with drawbacks:

- The "Script to Clipboard" feature will still not work.
- It's a quick fix that sidesteps the underlying issue of the interface being heavily entangled with the ADS.

**Conclusion:**
From this investigation it is apparent that Solution 2 is not a great solution and feels like a "quick fix". Also, Solution 1 demands a considerable amount of time and effort. Given that this issue is of low priority for Indirect Inelastic, I do not think a solution for this issue should be explored any further. Now I have a clearer understanding of the trade-offs and priorities regarding this issue, I am going to close it. If circumstances change or the issue becomes more critical in the future, it can always be revisited.

PR #36368 (2023-12-01): If StoreInADS False then set OutputWorkspace property optional
### Description of work
When StoreInADS is False, the output workspace from an algorithm tends to not be put into the ADS. Also, a workspace does not actually have a name until it is put into the ADS. Therefore, providing a value to the `OutputWorkspace` property of an algorithm should be optional when StoreInADS is False, because the output workspace name is not used when the output is not placed into the ADS.

Making this behavioural change will allow developers to decouple their interface code from the ADS, whilst still allowing [... cut]
closed 2023-12-04 by robertapplin