mantidproject/mantid #28669: Algorithm Dialog functions should be removed from Autocompletion

DanielMurphy22, 2020-05-12:
When converting the Python In Mantid Course, I have been removing the Algorithm Dialog references, but Algorithm Dialogs are still part of Autocompletion.

Type `LoadDia` into Workbenhc script editor and see that LoadDialog is recommended! 

Workbench 5.0
This should probably be fixed for 5.2, when MantidPlot is deprecated.

martyngigg, 2020-10-06:
We should do this now as the functions require MantidPlot.

They need to be removed from `mantid.simpleapi` and then they will be removed from the autocomplete.

martyngigg, 2020-10-06:
There are also some handwritten functions such as `def LoadDialog` that will also need to be removed and some unit tests will need updating to removing checking the functions exist.

PR #29674 (2020-10-21): Remove algorithm dialogs from simpleapi
Removes the algorithm dialogs FitDialog and LoadDialog from the simple api as they are deprecated. They will no longer appear in the autocompletion in the script editor.

**To test:**

Ensure neither `FitDialog` nor `LoadDialog` appear in the autocompletion in the script editor.

Fixes #28669 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http:/ [... cut]
closed 2020-10-21 by gvardany