mantidproject/mantid #28678: Access Sliceviewer / Instrument view from a script

DanielMurphy22, 2020-05-14:
When updating the Python in Mantid Course to be workbench foccussed.

You could in Plot, but cannot yet it Workbench open Instrument Viewer or Slice Viewer from a script.

Previous Example code for instrument viewer in Plot:
```
	# First get a handle to the view for a given workspace
	instrument_view = getInstrumentView("workspace-name")

	# Then retrieve a handle to the render tab
	render = instrument_view.getTab(InstrumentWidget.RENDER)

	# Set the colour map range
	render.setMinValue(10.)
	render.setMaxValue(50.) 

	# Raise the window
	instrument_view.show()

	# Change the colour map
	import os
        render.changeColorMap(os.path.join(config["colormaps.directory"],
                                                                       "BlackBodyRadiation.MAP"))

	# Set the colour map min/max together
	render.setRange(10.,50.)
	# Alter the binning range
	instrument_view.setBinRange(10000,15000)

	# Select a component that the view should zoom to
	# (Note: This is not case sensitive). Remember to give a valid component name.
	tree_tab = instrument_view.getTab(InstrumentWidget.TREE)
	tree_tab.selectComponentByName("low angle bank")

	# Load the instrument into a workspace
	import os
	inesPath = os.path.join(ConfigService.getInstrumentDirectory(),"INES_Definition.xml")
	LoadEmptyInstrument(Filename=inesPath,
	                    OutputWorkspace="INES")
	view = getInstrumentView("INES")

	# Other operations as above
	# ... 
	# ...

	# Show the window
	view.show()
```
https://www.mantidproject.org/Instrument_View_Control

Previous Example code for Sliceviewer in Plot:

```
    # Creating a multidimensional workspace - this will take a while and a lot of memory
    SXD23767 = Load(Filename='SXD23767.raw', LoadMonitors='Exclude')
    mdws=ConvertToMD( InputWorkspace=SXD23767, QDimensions="Q3D",
                        dEAnalysisMode="Elastic", QConversionScales="Q in A^-1",
                    LorentzCorrection='1', MinValues=[-15,-15,-15], MaxValues=[15,15,15],
                        SplitInto='2', SplitThreshold='50',MaxRecursionDepth='14' )

    # Plot a slice with all defaults applied
    svw = plotSlice(mdws, xydim=['Q_lab_x','Q_lab_y'], slicepoint=[0,0,5.8], colorscalelog=True) 

    # Specifiying the slice location
    # Move the slice point 
    svw.setSlicePoint('Q_lab_z', 5.9)

    #Plotting a line integration
    lv = svw.showLine(start=[-7, 6], end=[7, 6], width=0.1)

    # Overplotting a peaks workspace
    peaksws = Load('peaks_qLab.nxs')
    sv = svw.getSlicer()
    sv.setPeaksWorkspaces(['peaksws'])
```
https://www.mantidproject.org/SliceViewer_Python_Interface_Demo

DanielMurphy22, 2020-05-14:
The Python in Mantid course (Now moved to the user docs) should be updated after these features are merged.

martyngigg, 2021-03-04:
So the instrument view can now be access via a [script](https://docs.mantidproject.org/nightly/workbench/instrumentviewer.html#python-control).

How much appetite do we think there is for accessing the other stuff? You can use the widgets library to access the [SliceViewer](https://github.com/mantidproject/mantid/blob/23fb6a7963e1688ca77be72a61716c42c58e4c73/qt/python/mantidqt/widgets/sliceviewer/__init__.py#L13) and you could do similar things with the data table for now.

DanielMurphy22, 2021-03-04:
I'm tempted to say that the "other stuff" is maybe more of a power user feature that hasn't been specifically requested after moving from MantidPlot. If requested in the future then we can open a new issue, so happy for this to be closed.

@martyngigg  Do you think it's worth adding some of the instrument viewer scripting into the new Python In Mantid course?

martyngigg, 2021-03-04:
@DanielMurphy22 Yes I think that's worthwhile as an addition to the course.

Closing this issue as not enough appetite currently with available resources.

closed 2021-03-04 by martyngigg