mantidproject/mantid #28690: Step scan analysis interface, should it be ported to workbench?

NickDraper, 2020-05-15:
The step scan analysis interface wsa created for ORNL some time ago, but should it be ported to the workbench, does anyone use it?

NickDraper, 2020-05-15:
The usage stats:
```
id	type	name	internal	count	mantidVersion	application
3136	Interface	Step Scan Analysis		36	3.7	mantidplot
3928	Interface	Step Scan Analysis		13	3.8	mantidplot
4488	Interface	Step Scan Analysis		27	3.9	mantidplot
6841	Interface	Step Scan Analysis		17	3.10	mantidplot
8061	Interface	Step Scan Analysis		12	3.11	mantidplot
9101	Interface	Step Scan Analysis		53	3.12	mantidplot
10651	Interface	Step Scan Analysis		160	3.13	mantidplot
12516	Interface	Step Scan Analysis		35	4.0	mantidplot
13959	Interface	Step Scan Analysis		71	4.1	mantidplot
16054	Interface	Step Scan Analysis		11	4.2	mantidplot
18549	Interface	Step Scan Analysis		11	5.0	mantidplot
```
These numbers are so low they could just be us testing it.

AndreiSavici, 2020-05-15:
Yes. There are people who use it to align their samples

peterfpeterson, 2021-01-11:
This has already been done.

closed 2021-01-11 by peterfpeterson

AndreiSavici, 2021-01-11:
Closed by #29998