mantidproject/mantid #28776: Harmonise Masking Approaches, removing reversible masking

NickDraper, 2020-06-02:
The cunfusion caused by having two masking approaches was discussed at the TSC 02/06/2020, and it was agreed that we should harmonise on the more widely used non-reversible masking.  Please see the minutes for more details.

The tasks in this issue are:
1. Check with the ESS and ILL whether this will cause problems as they were not present at the meeting. (@gvardany , @SimonHeybrock please let me know if this will cause problems with your established workflows).
1. Adjust MaskDetectorsIf and MaskInstrument to clear the data after masking (update the docs too)
1. Remove the ClearMaskedDetectors algorithm.

gvardany, 2020-06-02:
This is exactly how it should be. If one needs to undo the masks, he has to start from raw data again. I'd just add a point that maybe in the dev-docs somewhere we should make it clear that at least any new C++ algorithm must respect both detector and bin masks, by checking their flags.

SimonHeybrock, 2020-06-03:
> The tasks in this issue are:
> 
>     1. Check with the ESS and ILL whether this will cause problems as they were not present at the meeting. (@gvardany , @SimonHeybrock please let me know if this will cause problems with your established workflows).
> 
>     2. Adjust MaskDetectorsIf and MaskInstrument to clear the data after masking (update the docs too)
> 
>     3. Remove the ClearMaskedDetectors algorithm.

I am not aware of problems with current workflows (and we will mostly be using scipp anyway).

To add to your list:
- LoadNexusProcessed
- Any other load algorithms for file types that support masks.

PR #29004 (2020-07-27): Remove reversible masking
**Description of work.**
1. MaskInstrument and MaskDetectorsIf changed to clear masked spectra after applying masking
1. Algorithm ClearMaskedSpectra Removed as its only purpose has been replaced in point 1 above
1. Updates to the workflow in ISIS SANS and AlignAndFocusPowder to replace MaskInstrument + ClearMaskedSpectra with just MaskInstrument.
1. Deprecated MaskInstrument as MaskDetectors should be used instead.

**To test:**

1. Unit tests and doc tests must pass (that covers most of it)
1. Code review

Fixes #28776

---

#### Reviewer ####

Please comment on the [... cut]
closed 2020-07-27 by gvardany