mantidproject/mantid #28788: Instrument view: Negative values not displayed

DanielMurphy22, 2020-06-04:
**Original reporter:** James Lord

### Expected behavior
The data (even negative) is represented by the colorbar
### Actual behavior
Any detectors with negative data values are greyed out, despite the colorbar trying to fit the correct range.
### Steps to reproduce the behavior
```
LoadMuonNexus(Filename=r'\\chronus\data\CHRONUS00009705.nxs', OutputWorkspace='CHRONUS00009705.nxs', DeadTimeTable='CHRONUS00009705.nxs_deadtime_table', DetectorGroupingTable='__notUsed')
RenameWorkspace(InputWorkspace='CHRONUS00009705.nxs', OutputWorkspace='Raw1')
 
LoadMuonNexus(Filename=r'\\chronus\data\CHRONUS00009706.nxs', OutputWorkspace='CHRONUS00009706.nxs', DeadTimeTable='CHRONUS00009706.nxs_deadtime_table', DetectorGroupingTable='__notUsed')
RenameWorkspace(InputWorkspace='CHRONUS00009706.nxs', OutputWorkspace='Raw2')
 
Minus(LHSWorkspace='Raw1', RHSWorkspace='Raw2', OutputWorkspace='Subtracted')
```
Right-click and Show Instrument on the `Subtracted` workspace

### Platforms affected
Found on Workbench 5.0.20200522.1617

NickDraper, 2020-06-04:
Looking into this:  The grey colouring is due to this code:
InstrumentRenderer.cpp line366:
``` C++
  for (size_t det = 0; det < counts.size(); ++det) {
    if (!isMasked(det)) {
      const double integratedValue(counts[det]);
      if (integratedValue > -1) {
        const auto &color = rgba[det];
        m_colors[det] =
            GLColor(qRed(color), qGreen(color), qBlue(color),
                    static_cast<int>(255 * (integratedValue / vmax)));
      } else
        m_colors[det] = invalidColor;
    } else {
      m_colors[det] = maskedColor;
    }
  }
```

NickDraper, 2020-06-04:
Just need to find out why -ve values are considered invalid?

NickDraper, 2020-06-04:
Is it only this code?
InstrumentActor.cpp line 437:
```C++
double InstrumentActor::getIntegratedCounts(size_t index) const {
  auto i = getWorkspaceIndex(index);
  if (i == INVALID_INDEX)
    return -1.0;
  return m_specIntegrs.at(i);
```

NickDraper, 2020-06-04:
As a quick and dirty brutal test: I commented out the check if the value was negative in the InstrumentRenderer about, and the result is that it can show negative values:
[image]

NickDraper, 2020-06-04:
The pick miniplot does not work though:
[image]

PR #28792 (2020-06-09): Instrument view handle invalid values better
**Description of work.**
Read the commit messages.

**Report to:** James Lord

**To test:**
``` python
from mantid.simpleapi import *
ws=LoadMuonNexus(Filename=r'CHRONUS00009705.nxs', OutputWorkspace='CHRONUS00009705.nxs', DeadTimeTable='CHRONUS00009705.nxs_deadtime_table', DetectorGroupingTable='__notUsed')
RenameWorkspace(InputWorkspace='CHRONUS00009705.nxs', OutputWorkspace='CHRONUS9705_raw_data MA')
 
LoadMuonNexus(Filename=r'CHRONUS00009706.nxs', OutputWorkspace='CHRONUS00009706.nxs', DeadTimeTable='CHRONUS00009706.nxs_deadtime_table', Detector [... cut]
closed 2020-06-09 by gvardany