mantidproject/mantid #28801: Colorfill plots setting z-axis scale and limits

darominski, 2020-06-08:
### Expected behavior

One should be able to select z-axis and change limits and scale from the `Figure Options` pop-up of a Colorfill plot when `Workspace2D` is plotted.

### Actual behavior

It is not possible to select the z-axis, and thus change limits nor scale from the Figure Options pop-up. 

### Steps to reproduce the behavior

In Mantid Workbench:
Load a `Workspace2D`, e.g. 164198.nxs from TrainingCourseData 
Right click on the workspace, select `Plot` and then `Colorfill`.
A pop-up with the plot should appear. 
Select `Figure Options`, try to select the z-axis.

### Platforms affected
Checked on Ubuntu 18.04 but I presume other platforms are also affected.

DanielMurphy22, 2020-06-10:
I believe the current behaviour is intentional. [The Z axis option in the Figure Options was added for 3D surface and wireframe plots](https://github.com/mantidproject/mantid/pull/28473). As such, the Z axis was disabled for 1D line plots and 2D Colorfill plots.
On a colorfill plot, the "Z-axis" is currently referred to as the Colorbar or Image. So in Figure Options, under the [Images tab](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html#image-plots) the scale and limits can be changed. These can also be changed from the [right-click menu](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html#click-menus) on the plot > Colorbar (Lin/Log), or [Double-click ](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html#click-menus)on the Colorbar Axis (Min/Max/Lin/Log).

gvardany, 2020-06-10:
@DanielMurphy22 This is exactly what is not working. In the image tab we don't get the workspace in the combo box, but for some reason we see the individual spectra.

gvardany, 2020-06-10:
[image]
[image]

alicerussell1, 2020-07-06:
@gvardany I can't seem to replicate this. What workspace have you used here?

gvardany, 2020-07-07:
This was the file, just rename to .nxs and load.
[d16_colorfill.nxs.zip](https://github.com/mantidproject/mantid/files/4882541/d16_colorfill.nxs.zip)

alicerussell1, 2020-07-07:
Ah okay I can see the issue now, I'll look into this

PR #28969 (2020-07-16): Fix figure options image tab settings for ws with numeric axis
**Description of work.**
This PR fixes an issue where colorfill plots done using pcolormesh of workspaces with an uneven numeric vertical axis are done per spectra so in the image tab in figure options each spectra was appearing as an image. This has been changed so that there is only one entry for each plot.

**To test:**
Load workspace `d16_colorfill` from the linked issue.
Do a colorfill plot
Open `Figure options` and go to the `Images` tab
In the select image drop down there should only be one image
Change some of the image set [... cut]
closed 2020-07-16 by gemmaguest