mantidproject/mantid #28809: Lazily create mantid.simpleapi function signatures

martyngigg, 2020-06-09:
On importing the `mantid.simpleapi` in Python we currently  [initialize every algorithm](https://github.com/mantidproject/mantid/blob/b2b8f9b29a649d22311224f7ea16fb8ab08fd904/Framework/PythonInterface/mantid/simpleapi.py#L1407). This is actually only required to [generate the function signature](https://github.com/mantidproject/mantid/blob/b2b8f9b29a649d22311224f7ea16fb8ab08fd904/Framework/PythonInterface/mantid/simpleapi.py#L107) and not for the function definition itself.

Explore implementing a lazy Signature object that only creates the actual function signature on demand when a user accesses it with help or the autocomplete in the GUI. A [basic prototype](https://gist.github.com/martyngigg/786a5797a60da1ed145e5253a2e8ab5a#file-simpleapi-py-lazysig-patch-L9) has been created and it appears to work well.

This will reduce import time, especially in a debug build, as it will reduce exception throwing and memory allocation drastically.

DavidFair, 2020-06-09:
Is there a way we can add some sort of system test which skips the lazy loading and forces Mantid to check all signatures are valid too? This way we can still be sure that when a user opens a dialog box it's not going to fall over

martyngigg, 2020-06-09:
We should probably add some test to verify these are okay. Just to be clear this isn't the do with the function parameters themselves. It's to do with what `help(Rebin)` produces or when you want to show the signature in the script editor for how to call it. The function parameters for all of `mantid.simpleapi` are `def f(*args, **kwargs)` but the dialogs are generated from the algorithm properties so never even go anywhere near inspecting either the function args or signature.

PR #28953 (2020-07-16): Lazily Generate Python Algorithm Function Signatures and DocStrings
**Description of work.**
Replaces the generation of docstrings and algorithm function signatures with lazy versions. This essentially means that they are only generated when they are requested. The intention is to reduce the amount of time and memory spent initialising all the python algorithms at startup. 

Note: Due to the way that the code editor completion works, this currenly does not affect the time to boot the workbench. All of the information that would normally be generated lazily is accessed on startup. The intention [... cut]
closed 2020-07-16 by gemmaguest