mantidproject/mantid #28827: Dialog Size assumes short WS name

DanielMurphy22, 2020-06-11:
**Original reporter:** Stephen King @smk78 
The Algorithm Dialog box currently assumes that Workspace names are less than a certain length and does not resize horizontally for particularly long ws names. We should not make this assumption.
### Expected behavior
The algorithm Dialog appears large enough to show all functionality 
### Actual behavior
In this case, the user had a rather long ws name, so the "Replace InputWorkspace name as Output WS name" button was not visible, as the Dialog did not horizontally resize. 
### Steps to reproduce the behavior
run `SupercalifragilisticexpialidociousSomethingQuiteAtrocious = CreateSampleWorkspace()`
Right-click on the workspace and select Rename. Can you see the 
[image] button? 
This issue may manifest itself differently for different screens / OSs.
Basically, the extent of the InputWorkspace box is going beyond the dialog edge.
### Platforms affected
Plot and WB all, noticed in Plot 5.0 on Windows

martyngigg, 2020-10-06:
@clayton-tom Did you make any headway with this? I think it ended up being way more complicated than expected?

smk78, 2020-10-06:
If that is the case, can you do anything that makes it more obvious that more functionality is available if you stretch the dialog?

clayton-tom, 2020-10-06:
I wasn't able to get it done before the end of the sprint, it turned out to be an issue in the code for the generic Algorithm dialogue creation but I couldn't find out where/why at the time. I've still got some notes on it, so happy to look into it again at some stage

martyngigg, 2020-10-06:
@clayton-tom Thanks. Could you put the notes here so that someone else could pick it up?

clayton-tom, 2020-10-07:
[image]

PR #30047 (2020-12-04): Set min length and adjust policy for algorthm combo box
OptionsPropertyWidget, use setSizeAdjustPolicy() and setMinimumContentsLength()

Fixes 28827

**Description of work.**

OptionsPropertyWidget, use setSizeAdjustPolicy() and setMinimumContentsLength()

**To test:**

See  

Fixes #28827

*This does not require release notes* because **small change?**

Before and after:
[image]

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does t [... cut]
closed 2020-12-04 by martyngigg