mantidproject/mantid #28847: SaveCanSAS1D no longer including the User File in SASprocess

smk78, 2020-06-12:
Testing WorkBench 5.0.20200611.1513 on W10/x64.

### Expected behavior
SaveCanSAS1D used to include the ISIS 'User File' in the <SASprocess> metadata field.

### Actual behavior
It is no longer doing so. The <term name="user_file"></term> field is empty.

### Steps to reproduce the behavior
Invoke the ISIS SANS interface.
Load a User File.
Load a Sample Scatter raw dataset.
Select 1D Reduction.
Select Save Options File or Both.
Select CanSAS (1D) output format.
Select the row of the table with the data.
Click Process Selected.
Examine the xml file that is created (scroll to the bottom). Example attached.
[test_rear_1D_4.4_16.5.zip](https://github.com/mantidproject/mantid/files/4772048/test_rear_1D_4.4_16.5.zip)

**Fixing this bug is a high priority for the ISIS SANS Group.**

Also see https://github.com/mantidproject/mantid/issues/28458

smk78, 2020-07-13:
From a recent Workbench Nightly:
```
		<SASprocess>
			<name>Mantid generated CanSAS1D XML</name>
			<date>13-Jul-2020 18:15:44</date>
			<term name="svn">5.0.20200625.1716</term>
			<term name="user_file"></term>
			<term name="sample_trans_run">9927</term>
			<term name="sample_direct_run">9947-add</term>
			<SASprocessnote/>
		</SASprocess>
```

From Mantid 3.6.0 (different input data):
```
		<SASprocess>
			<name>Mantid generated CanSAS1D XML</name>
			<date>30-Mar-2016 14:37:41</date>
			<term name="svn">3.6.0</term>
			<term name="user_file">K:/masks/USER_LOQ_154F_BernardoXPRESS_12mm_Changer_HAB.txt</term>
			<SASprocessnote/>
		</SASprocess>

```

smk78, 2020-07-13:
Also check that the user file is being written by SaveNxCanSAS.

smk78, 2020-09-18:
Looking at the testing I have performed over the last week or so using 5.0.20200909.1539 and 5.0.20200914.1537 I note that only the former seems to have populated the User File field in CanSAS1D xml files (I haven't checked the NXcanSAS)
```
			<date>11-Sep-2020 14:19:58</date>
			<term name="svn">5.0.20200909.1539</term>
			<term name="user_file">USER_LOQ_202A_M4_Team_LOQ_8mm_Changer_BOTH.txt</term>
```
whereas in the latter the field is empty
```
			<date>16-Sep-2020 16:21:14</date>
			<term name="svn">5.0.20200914.1537</term>
			<term name="user_file"></term>
```

I am attempting to check with the latest nightly, but it is possible something odd has happened here.

smk78, 2020-09-18:
See https://github.com/mantidproject/mantid/issues/28458

martyngigg, 2020-09-21:
@gemmaguest @DavidFair This one currently has the 5.1S5 milestone. Does it need to be in 5.1 now?

DavidFair, 2020-09-21:
This was completed but had a bug where it wasn't output for merged reductions, which can catch people out as the HAB and LAB files will have it.

#28458 should fix it along with a broken import that made unit tests always pass

martyngigg, 2020-09-21:
Thanks. I'll update the milestone.

martyngigg, 2020-09-21:
Sorry, clicked the wrong button :facepalm:

martyngigg, 2020-09-23:
As #29543 is now merged should this issue be considered closed or is there more work to do?

smk78, 2020-09-23:
I've been wanting to test @DavidFair 's handiwork, but haven't managed to get a Nightly to download the past two days! SourceForge is currently pushing the odd packet my way. Can I get back to you on this this afternoon?

DavidFair, 2020-09-23:
@smk78 We've been looking into alternatives because of this, your not the only one affected by it.
In the mean time you could download the installer direct from the build servers, under "Latest Successful Artefact"
https://builds.mantidproject.org/job/master_clean-windows/

smk78, 2020-09-23:
Testing Nightly 5.0.20200922.1715 this seems to be working. Also test the loading of the CanSAS1D (XML) and NXcanSAS (HDF5) output with this field in SasView 5.0.3. They load and display correctly.

Closing this issue.

closed 2020-09-23 by smk78