mantidproject/mantid #28855: Make Indirect Corrections UI work with new PaalmanPingsMonteCarloAbsorption algorithm

DannyHindson, 2020-06-16:
The Indirect Correction user interface has a tab called "Calculate Monte Carlo" that currently calls the CalculateMonteCarloAbsorption algorithm. This algorithm has been superseded by the PaalmanPingsMonteCarloAbsorption algorithm. This task is to update the UI to move over to the new algorithm.

As discussed with Spencer

AnthonyLim23, 2020-06-16:
can we have a chat about this?

AnthonyLim23, 2020-06-16:
Looks like the new alg has some new properties:
```
NumberOfWavelengthPoints
```

The old GUI has the following properties, but the new one does not:
```
SparseInstrument
NumberOfDetectorRows
NumberOfDetectorColumns
ContainerRadius
```

AnthonyLim23, 2020-06-22:
its ok to swap out

closed 2020-06-23 by martyngigg