mantidproject/mantid #28870: Flip Spectrum numbers on Colorfill if origin=upper

DanielMurphy22, 2020-06-17:
See #28811 for further explanation

### Expected behavior
Plotting a colorfill from a script should always show the correct axes numerals
### Actual behavior
If imshow is used, with out specifying `origin = 'lower'`, `upper` (the default for MPL) is used. As Mantid expects this to be set to `'lower'`, the Y-axis numbers are displayed in the reverse order and thus the plot is wrong. 

While documentation and script generation helps to ensure that users set `origin = 'lower'`
In the case that While documentation and script generation helps to ensure that users set `origin = 'upper'`, Mantid should flip the order of the Y (spectrum) axis on colorfill plots.
### Steps to reproduce the behavior
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
nrows, ncols = 5, 5
signal = np.arange(0,nrows*ncols)
x = np.arange(0,2*ncols+1,step=2)

data = CreateWorkspace(DataX=x, DataY=signal, NSpec=nrows,UnitX='TOF')
figC,axC = plt.subplots(subplot_kw={'projection':'mantid'})
c=axC.imshow(data)
axC.set_title('Imshow')
cbar=figC.colorbar(c)
plt.show()
```
### Platforms affected
All Workbench

NickDraper, 2020-07-13:
Or should we just force origin = 'lower' if it is not specified in the command?  This means we would not interfere if someone actually specifically said origin = 'upper'

DanielMurphy22, 2020-07-13:
If the user doesn't set origin specifically, us choosing origin = lower would be good. 

This issue is also for the case that origin is set to upper as currently the axis labels and the displayed image are in the reverse order! So correcting this (with the proposed flip) is also important.

NickDraper, 2020-07-13:
@DanielMurphy22 Could you split this, the default origin could be induction 2, but the other part is a bit too complex for induction and should just be put in the core team icebox for sprint 5.

DanielMurphy22, 2020-07-13:
Second issue split off: #28997

PR #29200 (2020-08-27): Flip spectrum numbers on Colorfill if origin=upper
**Description of work.**
This PR flips the spectrum numbers on colorfill plots (made through imshow) if origin=upper. It does this by setting the extent and making sure the subsequent world -> pixel transformation respects the flipped axis limits. This PR also makes imshow default to origin lower if no value is set.

**To test:**
The following script should correctly flip the spectrum numbers, you can compare with right clicking and selecting colorfill through the workspace widget.

```
# import mantid algorithms, numpy and matplotlib
from man [... cut]
closed 2020-08-27 by gemmaguest