mantidproject/mantid #28874: Imshow with LogNorm fails

DanielMurphy22, 2020-06-18:
The following script doesn't work on the current master but worked before this commit: https://github.com/mantidproject/mantid/pull/28677/commits/0e688e861ed19a1af564dbd064dd5f18fbd81e95

```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm

data = Load('MAR11060')

fig, axes = plt.subplots(subplot_kw={'projection':'mantid'})

# IMPORTANT to set origin to lower
c = axes.imshow(data, origin = 'lower', cmap='viridis', aspect='auto', norm=LogNorm())
axes.set_title('MAR11060')
cbar=fig.colorbar(c)
cbar.set_label('Counts ($\mu s$)$^{-1}$') #add text to colorbar
plt.show()
```

Error Trace:

```
ValueError: Data has no positive values, and therefore can not be log-scaled.
  File "<string>", line 17, in <module>
  File "/usr/lib/python3/dist-packages/matplotlib/figure.py", line 1864, in colorbar
    cb = cbar.colorbar_factory(cax, mappable, **kw)
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 1368, in colorbar_factory
    cb = Colorbar(cax, mappable, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 946, in __init__
    ColorbarBase.__init__(self, ax, **kw)
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 329, in __init__
    self.draw_all()
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 355, in draw_all
    self._config_axes(X, Y)
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 463, in _config_axes
    self.update_ticks()
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 380, in update_ticks
    ticks, ticklabels, offset_string = self._ticker()
  File "/usr/lib/python3/dist-packages/matplotlib/colorbar.py", line 623, in _ticker
    b = np.array(locator())
  File "/usr/lib/python3/dist-packages/matplotlib/ticker.py", line 2097, in __call__
    return self.tick_values(vmin, vmax)
  File "/usr/lib/python3/dist-packages/matplotlib/ticker.py", line 2123, in tick_values
    "Data has no positive values, and therefore can not be "
```

DanielMurphy22, 2020-06-18:
Latest confusion: In trying to find a simple compromise between 2D script generation and the simple imshow above, I found that this script works:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm

data = Load('MAR11060')

fig, axes = plt.subplots(subplot_kw={'projection':'mantid'})

# IMPORTANT to set origin to lower
cfill = axes.imshow(data, origin = 'lower', cmap='viridis', aspect='auto')
cfill.set_norm(LogNorm(vmin=0.0001))
axes.set_title('MAR11060')
#cbar = fig.colorbar(cfill, ax=[axes])
#cbar.set_label('Counts ($\mu s$)$^{-1}$') #add text to colorbar
plt.show()
```
BUT uncommenting the two cbar lines gives a script that fails with the same error trace as above.
BUT THEN adding a vmax to the LogNorm so the 11th line reads:
`cfill.set_norm(LogNorm(vmin=0.0001, vmax=3792.3352))` works!

NickDraper, 2020-06-18:
So I suspect it is the vmin that saves it, as Nan will be excluded that way

DanielMurphy22, 2020-06-18:
Maybe vmin and vmax both need to be provided for either to be obeyed, as the vmin alone, with the cbar lines included, fails.

martyngigg, 2020-06-19:
I think the problems here is to do with our `modest_image` type. I temporarily commented out [return mantid.plots.modest_image.imshow(axes, z, *args, **kwargs)](https://github.com/mantidproject/mantid/blob/484d482f2acb90cb01e1c68cc96a81c614813982/Framework/PythonInterface/mantid/plots/axesfunctions.py#L652) and replaced it with `return axes.imshow(z, *args, **kwargs)` and the plot works as expected.

Looking at several issues around this on the matplotlib list it would seem that matplotlib uses numpy masked arrays to handle invalid values like nan/inf - I think we are losing this mask. In the matplotlib Image class the [`set_data`](https://github.com/matplotlib/matplotlib/blob/09d8da28c44b7402f7b4225f75f480abbbb18386/lib/matplotlib/image.py#L693) method masks invalid values as one of the first things it does but it looks like our [override](https://github.com/mantidproject/mantid/blob/484d482f2acb90cb01e1c68cc96a81c614813982/Framework/PythonInterface/mantid/plots/modest_image/modest_image.py#L57) is not doing that. Changing that line to `self._A = cbook.safe_masked_invalid(A)` locally fixes the problem for me. Note that I don't think we need the copy as we are using it with a temporary array generated from the workspace.

PR #28889 (2020-06-19): stop imshow calls with log plots and invalid data failing
**Description of work.** 

for the attention of: @DanielMurphy22 

Made mosest_image handle invalid data like matplotlib image

**To test:**
1. open workbench, run the scripts, see two valid graphs on with log scale, the other linear.
1. check that the existing colourfill functionality in the workspace menu still works
``` python
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm

data = Load('MAR11060')

fig, axes = plt.subplots(subplot_kw={'projection':'mantid'})

# IMPORTANT to set or [... cut]
closed 2020-06-19 by gvardany