mantidproject/mantid #28891: ConvFit temperature correction does not work with negative values

Mantid-Matthew, 2020-06-19:
Currently the temperature correction on the ConvFit tab does not work with negative energy values.

AnthonyLim23, 2020-06-22:
the current form is `x*C/(T*(1-exp(-xC)))` so for x<0 the denominator goes to infinity and for x->0 the denominator tends to 0.

AnthonyLim23, 2020-06-23:
```
class Tcorrection(IFunction1D): # or IPeakFunction
    def init(self):
        self.declareParameter("T", 300.)
    def function1D(self, xvals):
        # Access current values during the fit
        c = 11.606/self.getParameterValue("T")
        return c*xvals/(1.-np.exp(-c*xvals))
    def functionDeriv1D(self, xvals, jacobian):
        c = 11.606/self.getParameterValue("T")
        for i, x in enumerate(xvals):
          value = 1.
          if x*x<2.:
              value = 1.
          else:
            e = np.exp(-c*x)
            num = c*e*(-c*x+e-1.)
            den = (e-1.)*(e-1.)
            value = num/den
          jacobian.set(i,0, value) # parameter at index 0
# Register with Mantid
FunctionFactory.subscribe(Tcorrection)
```

Can prove using L'Hopital's rule that in the limit x->0 the function tends to 1

AnthonyLim23, 2020-07-24:
Load any file from IRIS or OSIRIS into ConvFit
Add a temperature correction
Add a flat background
Set the fit to only work on spec 1 (for time)
Set start x<0 and end x>0
Do the fit
Then we get errors

PR #29056 (2020-08-11): Add ConvFit temperature correction function
**Description of work.**
This PR creates a temperature correction function which will be used in the ConvFit tab in place of the user defined exponential. This is necessary as the user defined exponential is undefined at Q=0, which means the direct convolution of a function using this correction results in errors (due to NaNs). This function prevents the undefined behavior by using f(Q)=1 when Q is zero. This is the limit of the original temperature correction function as Q->0. 

**To test:**

- Data to test each tab of the Indirect Data Analysis GUI [... cut]
closed 2020-08-11 by DavidFair