mantidproject/mantid #28914: Allow PaalmanPingMonteCarloAbsorption to run on elastic instruments

DannyHindson, 2020-06-24:
The PaalmanPingsMonteCarloAbsorption algorithm is intended to allow elastic (diffraction) datasets to be processed but at the moment it fails because of a bug in this piece of code which results in the algorithm requiring an EFixed parameter even for a diffraction instrument:
```
        self._sample_unit = self._sample_ws.getAxis(0).getUnit().unitID()
        if self._sample_unit == 'dSpacing':
            self._emode = 'Elastic'
        else:
            self._emode = str(self._sample_ws.getEMode())
        if self._emode == 'Indirect' or 'Direct':
            self._efixed = self._get_efixed()
```
The final if statement should instead be if self._emode == 'Indirect' or if self._emode == 'Direct' so that it evaluates to false for a diffraction instrument. At the moment the string constant 'Direct' isn't compared to self._efixed and is always evaluated as true.

There is also a more minor problem that some of the "No Efixed parameter found" error is suppressed when running with an elastic dataset and a cylinder shape. You instead get a "RuntimeError: '<=' not supported between instances of 'NoneType' and 'NoneType'" error that comes from some of the validation checks in validateInputs. These don't work because _setup has exited early and so self_container_radius is empty.

marshallmcdonnell, 2020-06-26:
@gvardany and @DannyHindson, sorry I didn't realize the [PaalmanPingsMonteCarlo](https://docs.mantidproject.org/nightly/algorithms/PaalmanPingsMonteCarloAbsorption-v1.html) algorithm existed until now when looking at this issue. This is great, been wanting this for a while for total scattering / diffraction, great to see the addition.

I know it is a separate issue but would you also be able to add this algorithm as an entry to the following table of a concept page that lists all the available correction algorithms and their capabilities?
https://docs.mantidproject.org/nightly/concepts/AbsorptionAndMultipleScattering.html#absorption-correction-algorithms-in-mantid-table

If not, I can look over the source and add it in at a later time.
Thanks for the work on this, very exciting!

DannyHindson, 2020-06-29:
@marshallmcdonnell, @gvardany I've created a separate issue (#28926) to update the concepts page with the PaalmanPingsMonteCarlo algorithm. There were a couple of other miscellaneous updates to that page also required so have included those also

closed 2020-07-09 by martyngigg