mantidproject/mantid #28938: Investigate improving unit conversions for diffraction instruments

DannyHindson, 2020-06-30:
Issue reported by Pascal Manuel, Helen Playford

Most diffraction instruments undergo a calibration that outputs three diffractometer constants (DIFA, DIFC, TZERO) that capture various systematic effects that may affect the measurements for a particular run (eg detector position offsets, sample absorption effects). These diffractometer constants can be used when converting the units of a workspace from TOF to d-spacing by running the algorithm AlignDetectors. At the moment when a conversion is run in the opposite direction these constants are not used (eg after focussing the "per bank" datasets may be converted from d spacing back to TOF).

Awaiting further details from Helen - but at this stage this task is to investigate:
1) how the conversion from d-spacing back to TOF might use the diffractometer constants
2) whether the unit conversion for diffraction could be consolidated into the ConvertUnits algorithm by storing either the diffractometer constants or the detector position offsets in the workspace. Do we need a separate algorithm such as AlignDetectors to do this?

The AlignDetectors algorithm is fairly well embedded in the ISIS powder diffraction scripts so any changes that are made could have knock on effects to various instruments

Harrietbrown, 2020-08-03:
Talking with Helen I think I should be looking into how the unit conversion in gettign detector geometry in foccused banks

DannyHindson, 2020-09-30:
Email from Helen on 30th July:

Hi both,

Here are a selection of files as promised. I believe I already sent you the necessary vanadium (and calibration) files, and that you can access the raw data on the network. Please shout if there are other things you need.

Thanks for looking into this, it’s much appreciated!

Best wishes,
Helen

[files_2.zip](https://github.com/mantidproject/mantid/files/5306763/files_2.zip)

DannyHindson, 2020-10-02:
Adding a couple of emails to Helen:

Hi Helen,

I’ve seen your out of office message so won’t keep sending you more emails but thought I’d write down a few more points while it’s fresh in my mind. Perhaps we can have a chat when you’re back:

1)	I had a look through your spreadsheet difc_calc.xlsx. The formula you are using for L_total calculation includes a component called L3 which is the detector height I think. We don’t include this component in the Mantid calculation – we just add L1 and L2 together as part of the DIFC calculation. Also I see you’ve got L3=1 (metre?) – this seemed quite large so just wanted to check it’s correct
2)	The spreadsheet calculates a DIFC value for each bank from an average L2 (I think). In Mantid we calculate a DIFC value per detector, then convert from TOF to d spacing per detector and then focus. Not sure if the difference matters. My instructions in my previous email below will just give you the per detector DIFC values
3)	There are 84 detectors that have the select column=0 in the Master_copy_of_grouping_file_with_essential_masks.cal file (these 84 are not the dummy detectors, there’s a separate list of ~40 dummy detectors). The 84 detectors are all in banks 3, 4 and 5 and Mantid ignores them. It looks as the average L2 and two theta values in your spreadsheet (column A) include these detectors

Thanks
Danny

From: Hindson, Danny (STFC,RAL,ISIS) 
Sent: 30 September 2020 17:27
To: Playford, Helen (STFC,RAL,ISIS) <helen.playford@stfc.ac.uk>
Subject: RE: Files as promised

Hi Helen,

Sorry it’s taken a while to get back to on this. We’ve just completed a release sprint where we do a lot of general testing in preparation for a Mantid release so not much time to look at problems or work on new features. Also Harriet has been reassigned for now onto a piece of work for the spectroscopy team so I’ll pick up the work on this conversion issue.

If I remember correctly one small part of this task was to let you know if it’s possible to output the diffractometer constants (DIFA, DIFC, TZERO) that Mantid has calculated. These are calculated by the AlignDetectors algorithm but the values aren’t stored in a workspace anywhere. If you insert this line into your python script immediately before the line that calls AlignDetectors you should get a table workspace output containing the “per detector” constants (see screenshot below):

ConvertDiffCal(OffsetsWorkspace="OutputOffsets",OutputWorkspace="DiffConstants")

If this isn’t what you were after then please let me know. I’ll move onto look at the main substance of the issue using the zip file you sent now

Thanks
Danny

DannyHindson, 2020-10-27:
Another email to Helen (Mon 26th Oct):

Hi Helen,

I had a look into the Mantid logic for generating a GSAS file following our call on Friday. The GSAS file has a header for each bank which contains a DIFC value for that bank. This is calculated at the moment without including any calibration information for the detectors. It just uses the L2 and theta values for each detector that are in the instrument definition file without using the calibration information calculated by CrossCorrelate\GetDetectorOffsets.

Do these “per bank” DIFC values get used by GSAS? If yes then perhaps this is the source of the problem.

I can incorporate the calibration offsets into the per bank DIFC calculations that are used to generate the GSAS file. I’m not sure if it matters but it seems like there are two approaches: a) just average the “per detector” DIFC values across the detectors in each bank, b) calculate average inputs to the DIFC formula (L2, theta, offset) and then calculate a “per bank” DIFCs from these. Do you know if it matters which is done? The logic to generate the GSAS headers is currently using b) but without incorporating the offsets.

Thanks
Danny

closed 2021-03-30 by martyngigg