mantidproject/mantid #28944: Investigate PredictFractionalPeaks and PredictSattelitePeaks not working in nightly

RichardWaiteSTFC, 2020-06-30:
PredictFractionalPeaks and PredictSattelite peaks create an empty table in nightly build - but this might not be a real issue (all the tests passed...?)

Harrietbrown, 2020-07-08:
Running the example script in the docs comes up with the right result, What steps did you do to get the empty table in the output?

RichardWaiteSTFC, 2020-09-29:
There appears to be a problem with PredictFractionalPeaks only on IDAaaS. The following works on my local build but not on IDAaaS
```
from mantid.simpleapi import *
LoadIsawPeaks(Filename='peaks.txt', OutputWorkspace='pp')
LoadIsawUB(InputWorkspace='pp', Filename='WISH00042366_roughUB.mat')
PredictFractionalPeaks(Peaks='pp', Hoffset='-0.2,0,0.2', Loffset='0', RequirePeaksOnDetector=False, FracPeaks='pfp')
```
with the following files  (will need to remove .txt extension from UB file)
[peaks.txt](https://github.com/mantidproject/mantid/files/5299823/peaks.txt)
[WISH00042366_roughUB.mat.txt](https://github.com/mantidproject/mantid/files/5299828/WISH00042366_roughUB.mat.txt)

RichardWaiteSTFC, 2020-10-01:
Doesn't seem to effect v5.1 on IDAaaS or other OS

closed 2020-10-01 by RichardWaiteSTFC