mantidproject/mantid #28970: QENS Sequential fit SpecMin and SpecMax not working

StephenSmith25, 2020-07-08:
If you run the QENSFitSequential example script from the .rst file on nightly you'll get errors

### Expected behavior
Should be able to set appropriate SpecMin and SpecMax values

### Actual behavior
An error is thrown and Workbench crashes

### Steps to reproduce the behavior
Run the script from the .rst file
```
# Load sample and resolution files
sample = Load('irs26176_graphite002_red.nxs')
resolution = Load('irs26173_graphite002_red.nxs')

# Set up algorithm parameters
function = """name=LinearBackground,A0=0,A1=0,ties=(A0=0.000000,A1=0.0);
(composite=Convolution,FixResolution=true,NumDeriv=true;
name=Resolution,Workspace=resolution,WorkspaceIndex=0;
name=Lorentzian,Amplitude=1,PeakCentre=0,FWHM=0.0175)"""
startX = -0.547608
endX = 0.543217
specMin = 0
specMax = sample.getNumberHistograms() - 1
convolve = True  # Convolve the fitted model components with the resolution
minimizer = "Levenberg-Marquardt"
maxIt = 500

# Run algorithm
result, params, fit_group = QENSFitSequential(InputWorkspace=sample,
                                                Function=function,
                                                PassWSIndexToFunction=True,
                                                StartX=startX, EndX=endX,
                                                SpecMin=specMin, SpecMax=specMax,
                                                ConvolveMembers=convolve,
                                                Minimizer=minimizer, MaxIterations=maxIt)
```

Either the .rst file is incorrect, or SpecMin and SpecMax are not dealt with appropriately by the algorithm.
### Platforms affected

SpencerHowells, 2020-07-08:
@StephenSmith25 There’s no problem selecting range from the gui.

StephenSmith25, 2020-07-08:
Okay thats good to know, this may just be due to out-of-date documentation.

AnthonyLim23, 2020-11-24:
@Harrietbrown is the on the to do list or the back burner?

Harrietbrown, 2020-11-30:
I'm not sure at what point it happened but the rst file had already been edited to remove the specMin specMax terms. I think @StephenSmith25 is right that it is a matter of out of date documentation so I'm closing this.

closed 2020-11-30 by Harrietbrown