mantidproject/mantid #28998: Reinstate detector and sample size/thickness information in CanSAS1D and NxCanSAS output

smk78, 2020-07-13:
SaveCanSAS1D used to write the sample thickness, beam shape, beam size and detector z distances to the metadata. See here for an example from Mantid 3.9.2:
```
		<SASsample>
			<ID>0.1-SDS-0.5% RM8W in 92% D2O_SANS</ID>
			<thickness unit="mm">1.000000</thickness>
		</SASsample>
		<SASinstrument>
			<name>LOQ</name>
			<SASsource>
				<radiation>Spallation Neutron Source</radiation>
			</SASsource>
			<SAScollimation>
				<aperture name="Disc">
					<size>
						<x unit="mm">12.000000</x>
						<y unit="mm">12.000000</y>
					</size>
				</aperture>
			</SAScollimation>
			<SASdetector>
				<name>main-detector-bank</name>
				<SDD unit="m">4.14902</SDD>
			</SASdetector>
			<SASdetector>
				<name>HAB</name>
				<SDD unit="m">0.579</SDD>
			</SASdetector>
		</SASinstrument>
		<SASprocess>
			<name>Mantid generated CanSAS1D XML</name>
			<date>23-May-2017 09:48:45</date>
			<term name="svn">3.9.2</term>
			<term name="user_file">K:/masks/USER_LOQ_171H_M4_Griffiths_12mm_Changer_MAIN.txt</term>
			<SASprocessnote/>
		</SASprocess>
		<SASnote>
		</SASnote>
```

At some point these got dropped! See here for something reduced in a recent Workbench Nightly:
```
		<SASsample>
			<ID>Glued_Pipes_7&amp;Ref_right_edge_Sans-z69_SANS</ID>
		</SASsample>
		<SASinstrument>
			<name>ZOOM</name>
			<SASsource>
				<radiation>Spallation Neutron Source</radiation>
			</SASsource>
			<SAScollimation/>
			<SASdetector>
				<name/>
			</SASdetector>
		</SASinstrument>
		<SASprocess>
			<name>Mantid generated CanSAS1D XML</name>
			<date>13-Jul-2020 18:15:44</date>
			<term name="svn">5.0.20200625.1716</term>
			<term name="user_file"></term>
			<term name="sample_trans_run">9927</term>
			<term name="sample_direct_run">9947-add</term>
			<SASprocessnote/>
		</SASprocess>
		<SASnote>
		</SASnote>
```

The missing user_file is ticketed as https://github.com/mantidproject/mantid/issues/28847 

The missing batch_file name is ticketed as https://github.com/mantidproject/mantid/issues/28458 

All this information should also be output by SaveNxCanSAS.

rprospero, 2020-08-03:
As I'm looking into this bug, the issue is further upstream than SaveCanSAS1D.  SaveCANSAS1D tests whether the workspace that has been passed to it has a `UserFile` property and, if not, drops it from the output file.  The question then becomes why the run for the workspace is missing this property.

smk78, 2020-08-03:
How odd, because you can't reduce data without this information!

smk78, 2022-02-22:
This needs addressing for data provenance and quality assurance reasons.

dehoni, 2023-02-26:
This issue goes a bit further and beyond the question if reduced files should contain instrument related information. The file structure allows explicitly to hold information about the SASsample. This can be any information on the sample from temperatures, sample orientation, time the data was taken, magnetic field etc. Often these include scanned variables, and with intention, the beam line is recording these states as it is the defining variable for the observed signal. Without this detail, the analysis of the data is futile, and it is utmost cumbersome to retrieve this information from the sample log in the raw data and combine it again with the reduced data. In the best case this operation is error-prone.
These are crucial information to make sense of the data and the lack of it might risk the scientific output from an experiment.

smk78, 2023-08-21:
Whilst testing MantidWorkbenchNightly-6.7.20230816.1558 to re-test https://github.com/mantidproject/mantid/issues/13801 for @rbauststfc I have noticed a small bug in the metadata reporting.

The commanded reduction was:
[image]

But the CanSAS1D output file reported:
[image]

ie, it did not report the can_trans_run as .n003

This should be fixed as part of this issue.

Also see https://github.com/mantidproject/mantid/issues/13801#issuecomment-1686271257

PR #35506 (2023-11-10): Save Additional Metadata to SANS Reduced Output
**Description of work**
Added back some metadata that had gone missing in the reduction output since Mantid 3.9.2.

**Purpose of work**
Retain metadata in the output needed for analysis down the road.

Requested by user @smk78 

**Summary of work**
- Adjusted SANSSave to make use of all of the saving parameters in SaveCanSAS1D
- Added those same properties to SaveNXCanSAS and use them in SANSSave
- Added loading for that metadata to LoadNXCanSAS

**To test:**

_Note: You may need to download a new version of the Training Course Data if you get er [... cut]
closed 2023-11-10 by SilkeSchomann