mantidproject/mantid #29025: Update MuonPreProcess to handle multiple time zeros

Mantid-Matthew, 2020-07-20:
MuonPreProcess shifts the input data in time depending on a time zero value. It needs to be updated so that it actually shifts each spectra separately in time depending on a list of time zero values.

Mantid-Matthew, 2020-09-01:
MuonPreProcess already has an optional. So the easiest way to handle potentially ragged spectra is probably to either specify this rebin or not.

AnthonyLim23, 2020-09-29:
https://docs.mantidproject.org/nightly/algorithms/LoadPSIMuonBin-v1.html#algm-loadpsimuonbin will produce data with multiple time zeros

AnthonyLim23, 2020-09-29:
https://github.com/mantidproject/mantid/blob/414a3eb4031057af54fcbf297b510876c5a8b31c/Framework/Muon/src/MuonPreProcess.cpp#L204

closed 2020-12-10 by DavidFair