mantidproject/mantid #29026: Update Muon Analysis GUI with new time zero options

Mantid-Matthew, 2020-07-20:
Update Muon Analysis GUI to handle multiple time zero analysis:

* Have options to specify a global override, global shift or table to specify the time zeros
* Have options to either rebin ragged workspaces and so a normal analysis or not to rebin ragged workspaces to allow each spectrum to be fitted individually.

AnthonyLim23, 2020-07-28:
what do you mean by global override and global shift?

Mantid-Matthew, 2020-07-28:
Global override means having the options to set all the time zero values to one value you specify
Global shift means shifting all the values by one value you specify

Mantid-Matthew, 2020-07-31:
Just a note that when the GUI is updated to handle multiple time zeros LoadPSIMuonBin should be updated to correctly shift each spectra by its own time zero.

Mantid-Matthew, 2020-08-28:
This can lead ragged spectra. There are two ways to deal with this. For most instances rebinning to consistent bins is an ok solution.  For some data however the correct solution is to keep the ragged workspace and then form groups of one detector each which can then be simultaneously fit. The GUI should be updated to be able to cope with both possibilities

AnthonyLim23, 2021-05-20:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

dataX = [0, 1, 2, 3, 4, 5] * 4
dataY = [100, 200, 300, 200, 100] * 4
input_workspace = CreateWorkspace(dataX, dataY, NSpec=4)

# Create a time zero table
#time_zero_table = CreateEmptyTableWorkspace()
#time_zero_table.addColumn("double", "time zero")
#[time_zero_table.addRow([i+2]) for i in range(4)]
time_zero_table = mtd["time_zero_table"]
output_workspace = MuonPreProcess(InputWorkspace=input_workspace,
                                  TimeZeroTable=time_zero_table, TimeMin=0.0)

max_time_bins = 0
max_spec = 0
tmp = mtd["output_workspace_1"]
for spec in range(tmp.getNumberHistograms()):
    if len(tmp.readX(spec))>max_time_bins:
        max_time_bins = len(tmp.readX(spec))
        max_spec = spec

    

outx = [x for x in tmp.readX(max_spec)]
outy = [y for y in tmp.readY(max_spec)]
for spec in range(0, tmp.getNumberHistograms()):
    if spec == max_spec:
        continue
    print(spec, tmp.readY(spec))
    for j in range(len(tmp.readY(spec))):
        outy[j] += tmp.readY(spec)[j]
    print("moo", tmp.readY(spec))
       
print(outy)
```

closed 2024-05-26 by github-actions[bot]