mantidproject/mantid #29027: Investigate simultaneous fits of individual spectra

Mantid-Matthew, 2020-07-20:
A simultaneous fit across each individual spectra of workspace is needed to help ascertain phase and time zero for each spectra. Liaise with scientists and investigate how this can most easily be done within the current muon framework.

AnthonyLim23, 2020-07-20:
is this on the asymmetry of a single detector (1 detector per group) or is it to the raw counts?

Mantid-Matthew, 2020-07-21:
I was left slightly uncertain in the meeting which of those two options they wanted to go for. As there are still some unknowns I've put this investigate issue in here so we don't forget about it.

Mantid-Matthew, 2020-09-01:
To add some more information here. When you have multiple time zeros applying these to the loaded spectra leads to a ragged workspace. There are two proposed ways to deal with this:
 * The first which will work for most data is to rebin the data to once again this common bins. This introudces a slight error on the order of the bin width but should be fine for most data.
 * The second is to create a group from each spectra individually and then simultaneously fit these. There are some slight changes needed to the GUI to allow this. There was also the suggestion of adding an ease of life feature and having a button which automatically creates one group per loaded period.

Matthew-Andrew, 2020-09-02:
There is also currently restrictions on the maximum number of groups/pairs that can be created at one time. This will probably need to be lifted for this approach to be viable.

closed 2021-04-27 by stale[bot]