mantidproject/mantid #29054: Workbench does not free memory when deleting workspaces

MathieuTillet, 2020-07-27:
### Expected behavior
When the user deletes a workspace, the memory should be freed and reusable by the rest of the system.

### Actual behavior
On Ubuntu, when deleting a workspace in Workbench, the associated memory is not freed system-wise.
Mantidplot seems unaffected.

### Steps to reproduce the behavior
**Warning note** : this will cause your computer to freeze, probably for a long time. Ensure nothing of importance is left unsaved before attempting it.

- Launch Workbench using `./bin/launch_mantidworkbench.sh`
- Create a workspace taking more than half your memory. Useful for this, 
```
N = 1
CreateSampleWorkspace(OutputWorkspace='ws',
                      NumBanks=N, 
                      BankPixelWidth=1000, 
                      Random=True, 
                      XMax=1, 
                      BinWidth=0.001)
```
uses N Go of memory.
- Delete this workspace. You can check used, free and available memory using `free` on a terminal, and verify that the available memory is back to its previous level.
- Launch another instance of Workbench, the same way, *without closing the first one*.
- Attempt to create a workspace taking more than the free memory, but less than the available.
- It will crash and not be able to load this data, while there is still memory available according to the system.
 
### Platforms affected
Tested on Ubuntu 18.04, on branch master of Workbench, with 64 Go of RAM.

martyngigg, 2020-07-28:
Further investigation points to `tcmalloc` behaving unexpectedly on Ubuntu 18.04. This behaviour is not observed on CentOS 7 where the memory is indeed returned to the OS for reuse by other processes.

Using the script from the original [tcmalloc report](https://github.com/mantidproject/documents/blob/master/Project-Management/TechnicalSteeringCommittee/reports/TCmalloc.md#memory-reporting), with minor updates for Python 3:

<details>
<summary>mem_report.py</summary>

```
# mem_report.py
import mantid
from mantid.kernel import MemoryStats
from mantid.api import FrameworkManager
from mantid.simpleapi import Load
import os

mem = MemoryStats()
res1 = mem.residentMem()/1024.
print("Resident Memory Usage (MB)", res1)
print()

filename = "MER06398.raw" # in systemtests/Data
ws = Load(Filename=filename)
mem.update()
res2 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res2)
print("Memory used (MB):",(res2-res1))
print()

print("Clearing all memory (workspaces etc)")
FrameworkManager.clear()
mem.update()
res3 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res3)
print("Memory change (MB)",(res3-res2))
print()
print("Loading file again...")
ws = Load(Filename=filename)
mem.update()
res4 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res4)
print("Memory change (MB):",(res4-res3))
print()

print("Clearing all memory (workspaces etc)")
FrameworkManager.clear()
mem.update()
res5 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res5)

print()
print("Overall change (MB):",(res5-res1))
```
</details>

<details>
<summary>Output</summary>

## Ubuntu 18.04

### Standard malloc
```
FrameworkManager-[Notice] Welcome to Mantid 5.0.20200724.1421
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Resident Memory Usage (MB) 220.3046875

Load-[Notice] Load started
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
Load-[Notice] Load successful, Duration 3.94 seconds
Resident Memory Usage (KB) 3330.1015625
Memory used (MB): 3109.796875
[... 90 lines cut ...]
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
Load-[Notice] Load successful, Duration 5.89 seconds
Resident Memory Usage (KB) 3735.63671875
Memory used (MB): 3472.4609375

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 301.265625
Memory change (MB) -3434.37109375

Loading file again...
Load-[Notice] Load started
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
Load-[Notice] Load successful, Duration 5.95 seconds
Resident Memory Usage (KB) 3726.609375
Memory change (MB): 3425.34375

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 292.26171875

Overall change (MB): 29.0859375
```

</details>

The output shows that tcmalloc behaves as expected on CentOS 7 but not on Ubuntu 18.04 - behaving differently to the versions originally tested.

martyngigg, 2020-07-28:
Newer [glibc versions](https://sourceware.org/legacy-ml/libc-alpha/2017-08/msg00010.html) now available on Ubuntu 18.04 have improvements to standard malloc that incorporate ideas from tcmalloc and other allocators. These ideas mitigate most of the performance issues that were originally seen with malloc in multi-threaded workloads. Unfortunately these updates are not available on CentOS 7.

One additional benefit of tcmalloc was this controlled release of memory back to the OS using the
`TCMALLOC_RELEASE_RATE` flag that we are now seeing not function correctly on Ubuntu 18.04. The standard allocators require this work to be done separately with `madvise` etc. We need some mechanism to allow the OS to reclaim memory for use in other processes, at least for longer running desktop applications such workbench. There is a question whether for systems like autoreduction we should not bother with this.

Another popular high-performance allocator is `jemalloc` used by [many high-profile projects](https://github.com/jemalloc/jemalloc/wiki/Background#adoption). It's aim are parallel to tcmalloc and it also provides similar capabilities for tuning behaviour.

The output of the mem_report.py script above on Ubuntu 18.04 with `jemalloc` used via `LD_PRELOAD` produces:
<details>
<summary>Ubuntu 18.04 with jemalloc</summary>
Resident Memory Usage (MB) 224.89453125

Load-[Notice] Load started
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
Load-[Notice] Load successful, Duration 3.91 seconds
Resident Memory Usage (KB) 3745.50390625
Memory used (MB): 3520.609375

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 229.0078125
Memory change (MB) -3516.49609375

Loading file again...
Load-[Notice] Load started
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
Load-[Notice] Load successful, Duration 3.92 seconds
Resident Memory Usage (KB) 3745.66015625
Memory change (MB): 3516.65234375

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 229.1953125

Overall change (MB): 4.30078125
</details>

and is more inline with how tcmalloc still functions on CentOS 7. An additional benefit appears to be that the release of large chunks of memory back to the OS, e.g. deleting large workspaces (>16G), with `jemalloc` appears to be far more efficient; tcmalloc would spend a large amount of time in a loop with madvise as confirmed by `strace`. Its allocation performance is very similar to tcmalloc.

martyngigg, 2020-07-28:
Concerning the deletion times of large workspaces:

<details>
<summary>mem_report_noload.py 16G workspace</summary>

```
# mem_report.py
import mantid
from mantid.kernel import MemoryStats
from mantid.api import FrameworkManager
from mantid.simpleapi import CreateSampleWorkspace
import os
import time

mem = MemoryStats()
res1 = mem.residentMem()/1024.
print("Resident Memory Usage (MB)", res1)
print()

N = 16
CreateSampleWorkspace(OutputWorkspace='ws',
                      NumBanks=N,
                      BankPixelWidth=1000,
                      Random=True,
                      XMax=1,
                      BinWidth=0.001)
mem.update()
res2 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res2)
print("Memory used (MB):",(res2-res1))
print()

print("Clearing all memory (workspaces etc)")
timestamp = time.time()
FrameworkManager.clear()
print(f"Time to delete {time.time()-timestamp}s")
mem.update()
res3 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res3)
print("Memory change (MB)",(res3-res2))
print()
print("Creating workspace again...")
CreateSampleWorkspace(OutputWorkspace='ws',
                      NumBanks=N, 
                      BankPixelWidth=1000, 
                      Random=True, 
                      XMax=1, 
                      BinWidth=0.001)
mem.update()
res4 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res4)
print("Memory change (MB):",(res4-res3))
print()

print("Clearing all memory (workspaces etc)")
FrameworkManager.clear()
mem.update()
res5 = mem.residentMem()/1024.
print("Resident Memory Usage (KB)",res5)
    
print()
print("Overall change (MB):",(res5-res1))
```

</details>

<details>
<summary>Ubuntu 18.04 with tcmalloc</summary>
FrameworkManager-[Notice] Welcome to Mantid 5.0.20200724.1421
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Resident Memory Usage (MB) 225.40625

CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 28.01 seconds
Resident Memory Usage (KB) 17227.44140625
Memory used (MB): 17002.03515625

Clearing all memory (workspaces etc)
Time to delete 162.95366287231445s
Resident Memory Usage (KB) 17227.44140625
Memory change (MB) 0.0

Creating workspace again...
CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 24.82 seconds
Resident Memory Usage (KB) 18492.5
Memory change (MB): 1265.05859375

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 18492.5

Overall change (MB): 18267.09375
</details>

<details>
<summary>Ubuntu 18.04 with jemalloc</summary>
FrameworkManager-[Notice] Welcome to Mantid 5.0.20200724.1421
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Resident Memory Usage (MB) 225.125

CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 30.69 seconds
Resident Memory Usage (KB) 17124.171875
Memory used (MB): 16899.046875

Clearing all memory (workspaces etc)
Time to delete 5.146293640136719s
Resident Memory Usage (KB) 227.59375
Memory change (MB) -16896.578125

Creating workspace again...
CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 30.81 seconds
Resident Memory Usage (KB) 17119.03515625
Memory change (MB): 16891.44140625

Clearing all memory (workspaces etc)
Resident Memory Usage (KB) 224.56640625

Overall change (MB): -0.55859375
</details>

The very slight increase in creation time for jemalloc is more than offset by the deletion time.

martyngigg, 2020-07-30:
@gvardany, there seems to be no real opposition to switching to `jemalloc`. I will outline the steps required:

1. Cleanup step: Remove references to `${TCMALLOC_LIBRARIES_LINKTIME}` across all CMakeList files. This will remove the ability to link to tcmalloc but with I primary usage being now through Python this point we only want to support usage via `LD_PRELOAD`.
2. Remove all tcmalloc references from [`MSVCSetup`](https://github.com/mantidproject/mantid/blob/207ca8a456547634e43e64b6c4e12698fad6dbfb/buildconfig/CMake/MSVCSetup.cmake#L66) as the allocator on Windows currently seems fine and tcmalloc support was never used.
3. Add a `FindJemalloc.cmake` much like [`FindTcmalloc`](https://github.com/mantidproject/mantid/blob/master/buildconfig/CMake/FindTcmalloc.cmake) and use it in [`LinuxSetup`](https://github.com/mantidproject/mantid/blob/master/buildconfig/CMake/LinuxSetup.cmake)
4. Update [`LinuxPackageScripts`](https://github.com/mantidproject/mantid/blob/207ca8a456547634e43e64b6c4e12698fad6dbfb/buildconfig/CMake/LinuxPackageScripts.cmake) and corresponding scripts to use jemalloc instead of tcmalloc. We can remove the `TCM_RELEASE` & `TCM_REPORT` variables.
5. Update spec and ns-control files in [dev-packages](https://github.com/mantidproject/mantid/tree/master/buildconfig/dev-packages) to install `jemalloc`. For [Ubuntu](https://packages.ubuntu.com/search?keywords=jemalloc&searchon=names&suite=bionic&section=all) `libjemalloc-dev` is the package and on [RedHat/CentOS](https://centos.pkgs.org/7/percona-x86_64/jemalloc-devel-3.6.0-1.el7.x86_64.rpm.html) it's `jemalloc-devel`.
6. Update [`Dockerfiles`](https://github.com/mantidproject/dockerfiles/) for docker-based CI agents - @DavidFair and I can help with this.
7. Request that non-docker-based agents have `jemalloc` installed. Best asked in #jenkins on Slack.

closed 2020-09-03 by LamarMoore