mantidproject/mantid #29059: Workbench: ISIS SANS Interface: 2D Reductions not possible

smk78, 2020-07-27:
Discovered testing mantidnightly-5.0.20200724.1421-win64 on W10/x64.

### Expected behavior
At the bottom of the ISIS SANS Reduction interface are two radio buttons, 1D (default) and 2D, which determine what type of reduction is to be executed when runs are processed. It is a binary choice.

### Actual behavior
When the 2D radio button is checked, a 1D reduction is still being executed.
THIS IS A BLOCKER; ISIS SANS CANNOT MOVE TO WORKBENCH UNTIL THIS IS FIXED.

### Steps to reproduce the behavior
Bring up the ISIS SANS Interface
Load a User File
Populate Sample Scatter, Sample Transmission, and Sample Direct fields in the table
Select Reduction 2D
Select Save Option Memory
Highlight the row of the table with the runs
Click Process Selected
Look for a workspace with *\_2D\_* in its name.

### Platforms affected
Windows at least. Probably all.

gemmaguest, 2020-07-29:
@smk78 This seems to be working for me. Could it be specific to the runs you're using? I'm using the LOQ demo in the sample data.

smk78, 2020-07-29:
On further investigation, I realised that the batch files I had been testing were all specifying user files. Thus this issue is actually a duplicate of that noted in https://github.com/mantidproject/mantid/issues/27300. Sorry!

Am closing this ticket.

closed 2020-07-29 by smk78