mantidproject/mantid #29061: import mslice.cli Hard Crash

DanielMurphy22, 2020-07-28:
**Original reporter:** Tatiana / Duc

### Expected behavior
Mslice functionality in a script should work seemlessly: https://mantidproject.github.io/mslice/cli.html
### Actual behavior
Crash when importing mslice
### Steps to reproduce the behavior
Open a workbench package on any OS
Run in script editor `import mslice.cli as mc`
### Platforms affected
Workbench (4.2,5.0,Nightly) on All OS I think [This does not affect Plot]

DanielMurphy22, 2020-07-28:
Command Line outputs: 

Mantid Version: 5.0.20200728.716
```
/Applications/MantidWorkbench.app/Contents/MacOS/MantidWorkbench: line 21: 72290 Segmentation fault: 11  env PYTHONHOME=${PYTHONHOME} env PYTHONPATH=${INSTALLDIR}:${PYTHONPATH} env QT_PLUGIN_PATH=$PLUGIN_DIR env QTWEBENGINE_DISABLE_SANDBOX=1 ${PYTHONHOME}/bin/python ${INSTALLDIR}/workbench-script $*
qt.qpa.fonts: Populating font family aliases took 171 ms. Replace uses of missing font family ".SF NS Text" with one that exists to avoid this cost. 
```
**I think the fonts lines at the end are NOT relevant**

Another attempt on the same version:
```
/Applications/MantidWorkbench.app/Contents/MacOS/MantidWorkbench: line 21: 72275 Segmentation fault: 11  env PYTHONHOME=${PYTHONHOME} env PYTHONPATH=${INSTALLDIR}:${PYTHONPATH} env QT_PLUGIN_PATH=$PLUGIN_DIR env QTWEBENGINE_DISABLE_SANDBOX=1 ${PYTHONHOME}/bin/python ${INSTALLDIR}/workbench-script $*
qt.qpa.fonts: Populating font family aliases took 167 ms. Replace uses of missing font family ".SF NS Text" with one that exists to avoid this cost. 
```

Workbench Version: 5.0.20200529.1413
```
/Applications/MantidWorkbenchNightly.app/Contents/MacOS/MantidWorkbench: line 21: 72204 Segmentation fault: 11  env PYTHONHOME=${PYTHONHOME} env PYTHONPATH=${INSTALLDIR}:${PYTHONPATH} env QT_PLUGIN_PATH=$PLUGIN_DIR env QTWEBENGINE_DISABLE_SANDBOX=1 ${PYTHONHOME}/bin/python ${INSTALLDIR}/workbench-script $*
```

martyngigg, 2020-07-31:
It looks like this is do with the custom algorithm registration inside [mslice](https://github.com/mantidproject/mslice/blob/966f05885864d06f09636e732d00ed2c6ad4275c/mslice/util/mantid/init_mantid.py#L13)

It can be replicated by placing this

```python
from mantid.kernel import *
from mantid.api import *

class HelloWorld(PythonAlgorithm):

    def PyInit(self):
        # Declare properties
        pass

    def PyExec(self):
        # Run the algorithm
        pass

class HelloWorld2(PythonAlgorithm):

    def PyInit(self):
        # Declare properties
        pass

    def PyExec(self):
        # Run the algorithm
        pass

# Register algorithm with Mantid
AlgorithmFactory.subscribe(HelloWorld)
AlgorithmFactory.subscribe(HelloWorld2)
```

in a file called `helloworld.py`, putting this in the `bin` directory and in workbench type `import helloworld`. It should crash.

The issue is most likely the updating of the algorithm tree when the new algorithms on subscribed. The new subscriptions come from a separate thread and kill the application when trying to update the GUI from a separate thread.

PR #29242 (2020-08-28): Pull in fix for mslice cli in workbench
**Description of work.**

Pulls in fix in mslice so that scripts in workbench don't randomly crash.

**To test:**

The following script should no longer crash:

```python
import mslice.cli as mc
import mslice.plotting.pyplot as plt

ws = mc.Load(Filename='MARIReductionAutoEi.nxs', OutputWorkspace='MARIReductionAutoEi')

fig = plt.gcf()
ax = fig.add_subplot(111, projection="mslice")
slice_ws = mc.Slice(ws, Axis1="|Q|,0.27356,11.0351,0.04671", Axis2="DeltaE,-30.0,58.2,0.3", NormToOne=False)

mesh = ax.pcolormesh(slice_ws, cmap="viridis")
mesh.set_clim(0.0, [... cut]
closed 2020-08-28 by DavidFair