mantidproject/mantid #29078: Workbench: SANS Interface: Transmission Use Fit checkbox being cleared on Process when set manually

smk78, 2020-07-30:
Testing Workbench 5.0.20200724.1421 on W10/x64.

### Expected behavior
In the ISIS SANS Interface, under Settings > Adjustment > Transmission > Fit is a checkbox Use Fit.
This (along with the fit type and wavelength range for the fit) can be set or cleared in the User File with, for example, FIT/TRANS/POLYNOMIAL3 2.2 10.0 or FIT/TRANS/OFF, respectively. But it should obviously _also_ be possible to set these manually.

### Actual behavior
The transmission fitting functionality is working correctly when set/cleared from a User File. But, when it is set manually in the interface, and the User exits Settings to Runs in order to Process data, something is clearing the Use Fit checkbox, so you only ever get unfitted transmissions!
[image]
[image]

### Steps to reproduce the behavior
Run Workbench
Activate the ISIS SANS Interface
Load a valid User File that does not use transmission fitting (LOQ example attached)
[USER_LOQ_194A_M4_Team_LOQ_8mm_Changer_MAIN_LIN.txt](https://github.com/mantidproject/mantid/files/5000032/USER_LOQ_194A_M4_Team_LOQ_8mm_Changer_MAIN_LIN.txt)
Populate a row in the table with at least a Sample Scatter, Sample Transmission and Sample Direct runs
Select the row and Process
A 1D reduction should be performed resulting in a 1D I(Q) vs Q and a folder with 2 unfitted transmissions.
Now go to Settings > Adjustment > Transmission > Fit, check Use Fit, set Fit Type to Polynomial, Set Poly Order to 3, check Use Custom Wavelength and set 2.2  and 10.0 Angs.
Return to Runs, and re-Process the data.
You should now get 4 transmissions, the 2 unfitted and 2 fits. But you don't.
If you return to Settings > Adjustment > Transmission > Fit you will find Use Fit is now unchecked.

### Note
I was not specifying an individual User File in the table, so this is not something covered by https://github.com/mantidproject/mantid/issues/27300.

### Platforms affected
Presumably all.

gemmaguest, 2020-07-30:
@smk78 I get an error about `DIRECT_LOQ_MAIN_143B.txt` not being found - where can I find that file (and presumably the others mentioned in the user file)?

smk78, 2020-07-30:
Sorry, I assumed you'd have those somewhere. Attached.
[mantid-files.zip](https://github.com/mantidproject/mantid/files/5000214/mantid-files.zip)

gemmaguest, 2020-08-03:
We agreed this is part of #27300 after all so closing as duplicate

closed 2020-08-03 by gemmaguest