mantidproject/mantid #29085: Mantid not starting on Workbench

AnthonyLim23, 2020-07-30:
Workbench does not open if the user OS has an asian language. 

From Koji:

Looks like workbench does not open on Asian OS. At some point we should fix it.

Following up my comment on the Asian language support of the latest Mantid Workbench/Plot, I have installed them on my laptop. Unfortunately both Workbench and Plot didn't work in my environment with the Japanese typing option installed. As for the Workbench, it just doesn't start, and after a little while waiting to start the software, the error window shows up (with the sad Mantid mascot). I don't know what went wrong. On the other hand I can see Mantid Plot is doing something to start (according to the process messsage), but while it's "Loading code editing widget", it crushes and stops. For now I'll keep using ver. 4.1.

NickDraper, 2020-07-30:
3 day timeboxed investigation

martyngigg, 2020-08-04:
I created a Windows VM with Hyper-V manager, installed the Japanese language packs and set Japanese as the default language. Workbench and MantidPlot both started for me. I've sent an email to Koji asking for more information to try and help us diagnose this.

martyngigg, 2020-08-07:
It would appear that the issue on the user's machine was nothing to do with the language options but rather the switch to Python 3.

<details>
<summary>Result of `mantid.simpleapi` import on command line<summary>
```
In [1]: import mantid.simpleapi as sapi
FrameworkManager-[Notice] Welcome to Mantid 5.0.0
FrameworkManager-[Notice] Please cite:
http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release:
http://dx.doi.org/10.5286/Software/Mantid5.0
Python-[Warning] Failed to load plugin
C:/Users/Public/Mantid/ScriptRepository/muon/Quantum/Algorithms_Autoload/quantumtabledriven.py.
Error: Traceback (most recent call last):
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 195, in
load_from_file
     name, module = load_plugin(filepath)
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 214, in
load_plugin
     module = loader.run()
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 66, in run
     return SourceFileLoader(name, pathname).load_module()
   File "<frozen importlib._bootstrap_external>", line 462, in
_check_name_wrapper
   File "<frozen importlib._bootstrap_external>", line 962, in load_module
   File "<frozen importlib._bootstrap_external>", line 787, in load_module
   File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
   File "<frozen importlib._bootstrap>", line 702, in _load
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 779, in exec_module
   File "<frozen importlib._bootstrap_external>", line 916, in get_code
   File "<frozen importlib._bootstrap_external>", line 846, in
source_to_code
   File "<frozen importlib._bootstrap>", line 219, in
_call_with_frames_removed
   File
"C:/Users/Public/Mantid/ScriptRepository/muon/Quantum/Algorithms_Autoload/quantumtabledriven.py",
line 1059
     print "examining param group ",g
           ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean
print("examining param group ",g)?

Python-[Warning] Failed to load plugin
C:/Users/Public/Mantid/ScriptRepository/muon/Quantum/Algorithms_Autoload/quantumtabletools.py.
Error: Traceback (most recent call last):
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 195, in
load_from_file
     name, module = load_plugin(filepath)
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 214, in
load_plugin
     module = loader.run()
   File "C:\MantidInstall\bin\mantid\kernel\plugins.py", line 66, in run
     return SourceFileLoader(name, pathname).load_module()
   File "<frozen importlib._bootstrap_external>", line 462, in
_check_name_wrapper
   File "<frozen importlib._bootstrap_external>", line 962, in load_module
   File "<frozen importlib._bootstrap_external>", line 787, in load_module
   File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
   File "<frozen importlib._bootstrap>", line 702, in _load
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 779, in exec_module
   File "<frozen importlib._bootstrap_external>", line 916, in get_code
   File "<frozen importlib._bootstrap_external>", line 846, in
source_to_code
   File "<frozen importlib._bootstrap>", line 219, in
_call_with_frames_removed
   File
"C:/Users/Public/Mantid/ScriptRepository/muon/Quantum/Algorithms_Autoload/quantumtabletools.py",
line 740
     print "pops=",pops
[... 69 lines cut ...]
     123                 # None is not quite accurate here, but we are
reproducing the
     124                 # behavior found in the C++ code for SimpleAPI.
--> 125                 parameters.append(Parameter(name,
pos_or_keyword, default=None))
     126         return Signature(parameters)
     127     else:

C:\MantidInstall\bin\lib\inspect.py in __init__(self, name, kind,
default, annotation)
    2511
    2512         if not name.isidentifier():
-> 2513             raise ValueError('{!r} is not a valid parameter
name'.format(name))
    2514
    2515         self._name = name

ValueError: 'Calculation type' is not a valid parameter name
```

</details>

This shows that some code in the script repository is still using Python 2 print syntax and the framework is now choking on it. Commenting out the `user.python.plugins.directories` key in `Mantid.user.properties` stops this being loaded and then both Workbench and MantidPlot start.

There is code in the framework to trap errors on Python plugin loading so it is not quite catching this for some reason. I'm closing this issue as there is no problem starting Mantid on an Asian-language OS. We can open a separate issue to more robustly handled syntax errors in plugins.

closed 2020-08-07 by martyngigg