mantidproject/mantid #29087: Workbench sliceviewer ellipsoid background shell not plotted correctly

RichardWaiteSTFC, 2020-07-30:
The inner radius of the ellipsoid background shell is not plotted correctly in workbench sliceviewer. A comparison of mantidplot (left) and workbench (right) is shown below

[image]

MantidPlot is displaying an background inner radius very similar to the peak radius as it should be, but the workbench sliceviewer displays a much larger inner radius.

### Steps to reproduce the behavior

### Platforms affected

martyngigg, 2021-01-11:
@RichardWaiteSTFC Would you have a script that could reproduce this?

RichardWaiteSTFC, 2021-01-11:
@martyngigg 
Note that the background inner radius is the same as the peak radius - but not on the sliceviewer plot
```
Load(Filename='SXD23767.raw', OutputWorkspace='SXD23767', LoaderName='LoadRaw', LoaderVersion='3')
ConvertToDiffractionMDWorkspace(InputWorkspace='SXD23767', OutputWorkspace='SXD23767MD', OneEventPerBin='0')
FindSXPeaks(InputWorkspace='SXD23767', PeakFindingStrategy='AllPeaks', AbsoluteBackground=2000, 
    ResolutionStrategy='AbsoluteResolution', XResolution=200, PhiResolution=3, TwoThetaResolution=3, 
    OutputWorkspace='peaks')
IntegratePeaksMD(InputWorkspace='SXD23767MD', PeakRadius=0.35, BackgroundInnerRadius=0.35,
    BackgroundOuterRadius=0.5, Ellipsoid=True, PeaksWorkspace='peaks', OutputWorkspace='peaks_int', 
    UseOnePercentBackgroundCorrection=False)
```

RichardWaiteSTFC, 2021-01-25:
Fixed by PR #30403

PR #30403 (2021-01-21): Fix plotting of elliptical bg shell in sliceviewer
**Description of work.**

The elliptical shell is created by an affine transformation of a unit circle - therefore the inner radius for the shell (before transform) needs to be determined from fractional thickness of shell (relative to outer background radius) not the absolute thickness (defined as the difference between the major radii of the inner and outer background ellipsoids).

**To test:**

Follow instructions in original issue (feel free to delete all but a couple of peaks to speed this up). In IntegratePeaksMD the background inner rad [... cut]
closed 2021-01-25 by RichardWaiteSTFC