mantidproject/mantid #29101: Investigate IntegrateEllipsoid inaccurate integration for weak peaks

RichardWaiteSTFC, 2020-08-03:
TOPAZ scientists have noticed IntegrateEllipsoids can give anomalous results for weak peaks - have compared to IntegratePEaksMD and think that there might be an issue - but haven't identified where. Perhaps a good starting place would be to have better tests with a better way of generating fake peaks of known covariance  (with/without counts = errors = 1) - for example see recent changes to IntegratePEaksMD tests.

DannyHindson, 2020-11-23:
Issue raised by SNS

RichardWaiteSTFC, 2020-11-23:
[image]
Issue with integrateEllipsoids not integrating weak peaks

DannyHindson, 2021-05-04:
Closing because we're not using IntegrateEllipsoids at ISIS and this was possibly fixed in #31049

closed 2021-05-04 by DannyHindson