mantidproject/mantid #29118: Script Editor Font Size inconsistent MacOS

DanielMurphy22, 2020-08-04:
**Original reporter:** [Malcolm on Forum](https://forum.mantidproject.org/t/different-tabs-have-different-font-size-in-workbench-editor-mac-os/674)

### Expected behavior
Font size is the same for all tabs on the script editor
### Actual behavior
Font size on the first tab is smaller than tab 2,3,4,... which all have the same font size.
### Steps to reproduce the behavior
On MacOS open a few tabs on the script editor by clicking the `+` button.
Notice that the font of the first tab was smaller than the others
You can scroll or use Cmd and (+/-) to make the font size smaller or larger: all tabs scale by the same amount. When tab 2 (+3,4 etc) are at the correct size, tab 1 is illegibly small.
### Platforms affected
Workbench MacOS, presumably all seen on 5.0

martyngigg, 2023-01-23:
@gemmaguest This has been rolling over many releases. Would you be happy for this to be in 6.6 as the fix is small and localized so easy to verify.

mantid-builder, 2023-01-24:
Fixed by #35060

PR #35060 (2023-01-23): Fix inconsistent fonts across script editor tabs on macOS
**Description of work.**

Removes ancient override of QScintilla2 zoom function that artificially inflated the zoom level causing strange inconsistency when creating new tabs and trying to set the same level across then all. We now set the font so the
default is fine and it's better to have consistent behaviour with how zooming works.

**To test:**

Please follow the instructions in the linked issue.

Fixes #29118

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAP [... cut]
closed 2023-01-24 by mantid-builder