mantidproject/mantid #29142: New dual minimzer

AnthonyLim23, 2020-08-06:
We want to create a new minimizer for fitting complex functions. This new minimizer will have 2 sets of parameters (A and B), within each iteration it will do a fit with B fixed then a fit with A fixed, allowing the solution to move towards the minima. As the diagram shows it should work similar to a conjugate gradient method, except we can only move parallel to the axis. 

We would need to add two new properties to fit:
1. A list of the parameters (A) to fix. We can then work out the other list (B) as all parameters must be in either A or B
2.  A minimizer to use within this one (don't allow this one to be selected).

We should also make sure that the minimizer complains if the list of parameters to fix is missing.

[double_min.pdf](https://github.com/mantidproject/mantid/files/5034222/double_min.pdf)

AnthonyLim23, 2021-04-28:
@SilkeSchomann this is from David and Duc

StephenSmith25, 2021-06-24:
I've had a brief think about this. This procedure you describe above is just coordinate descent https://en.wikipedia.org/wiki/Coordinate_descent , which has quite a few issues when the function isn't convex (as per the image). I don't know if this requires a new minimiser as most of it can be done in a script:

e.g with scipy https://gist.github.com/StephenSmith25/10a75d4524fdf16fa0b6889d2c1eca4c
which gives

[image]

It would be a bit more complicated defining the directions for multi parameters, but should still be scriptable, giving that the mantid functions have a python interface 

```
x = 1
A0 = 2
A1 = 3
func = LinearBackground()
y = func(x, A0, A1)
print(y)
```

AnthonyLim23, 2021-06-24:
This was wanted for crystal field, which is buried inside Mantid's code base. I believe the aim is  to minimize across physical and modelling parameters in turn. 

@SilkeSchomann do you think it can be done in a script or does it need to be in the crystal filed code?

SilkeSchomann, 2021-06-24:
I doubt that this can be done by a script but I will keep the possibility in mind when working on the design for this.

SilkeSchomann, 2021-12-14:
**Design update**
We decided to use a slightly different approach than described above: 
- The dual fitting process is not implemented on the level of Mantid minimizers but as part of the Crystal Field Python interface. 
- In the first step only free field parameters are fitted, in the second step only peak parameters.
- The number of iterations within each step can be configured individually just like the overall number of iterations. 
- Two variants of the process are available. One uses Mantid fitting in both steps, the other one uses scipy.optimize.minimize in the first step and Mantid fitting in the second. Furthermore, scipy.optimize.minimize is applied to all free field parameters sequentially whereas the variant applying Mantid fitting tries to fit all free field parameters at the same time.

closed 2021-12-20 by martyngigg