mantidproject/mantid #29145: Mantid crashes when cancel is clicked during MaxEnt on Frequency domain analysis

tolu28-coder, 2020-08-06:
### Expected behavior
When cancel is pressed on frequency domain analysis gui when calculating MaxEnt, calculation should stop.

### Actual behavior
When cancel is pressed on frequency domain analysis gui when calculating MaxEnt, Mantis crashes.

### Steps to reproduce the behavior
1. Open Frequency Domain Analysis gui.
2. Load Appropriate data.
3. Calculate MaxEnt then click on cancel , mantid crashes.

### Platforms affected
Confirmed on Windows, assumed to work on other platforms as well.

AnthonyLim23, 2020-09-18:
for data use MUSR 62260
On the transform tab change drop down to MaxEnt
Change the number of data points to a high number (this will make it slow, so better chance to cancel)

AnthonyLim23, 2020-09-18:
Its definitely in the GUI, the cancel works ok when running the MuonMaxEnt algorithm

PR #29531 (2020-09-22): Fixes Crash Caused By Pressing Cancel In FDA During MaxEnt
**Description of work.**
Fixed a crash which was caused by pressing cancel during execution of the MuonMaxEnt algorithm from the Frequency Domain Analysis GUI.

**To test:**
1. Open the FDA interface from **Interfaces** -> **Muon** -> **Frequency Domain Analysis**
2. Load in a dataset such as MUSR **62260**
3. Switch to the **Transform** tab
4. Change the drop down box from **FFT** to **MaxEnt**
5. Change the **Number of data points** in Advanced Options to a large number (this is so you have time to press cancel!)
6. Click **Calculate [... cut]
closed 2020-09-22 by martyngigg