mantidproject/mantid #29162: Elemental Analysis multiple crash minor peaks

DanielMurphy22, 2020-08-10:
**Original reporter:** Adrian Hillier
@AnthonyLim23 mentioned this may be related to #28392 

It is very easy to produce repeat error reporters, so you have to terminate Mantid workbench.
A realistic workflow for this crash involves loading data and plotting, but simply opening the interface and ticking then unticking the minor ticks box, produces these error reporters. The interface struggles to remove minor peaks cleanly.

### Steps to reproduce the behavior
Short:
- Open Elemental Analysis Interface
- Tick the Minor peaks box
- Untick it.

More realistic:
- Open Elemental Analysis Interface
- Put `Sundials` into your Data search directories in MUD (Contact myself or @AnthonyLim23)
- Enter `2695` and click Load.
- Plot detector GE1
- Select minor peaks
- Deselect minor ticks (or infact the major ticks!)

### Platforms affected
Workbench All, found on Windows 5.0.20200805.613

AnthonyLim23, 2020-09-10:
We should fix this for this release. It looks like there are few issues:
When unticking major/minor peaks. It picks up lines that don't exist. This crash can happen without selecting any elements or a plot
Right clicking Cu and unticking anythin after the first instances of a repeated line e.g. L(3d->2p): will cause an error. We must have an issue where the spaces are being removed so the keys are identical (we used spaces to make the keys different)

AnthonyLim23, 2020-09-10:
looks like its any of the peak boxes

AnthonyLim23, 2020-09-10:
some of the lines (the ones with the same transition names) are not being plotted

PR #29418 (2020-09-15): Fixed Bug with unselecting peak checkbox on elemental analysis
**Description of work.**
Fixed a Bug that caused the elemental analysis gui to crash with any of the peak checkbox was unselected.

**Report to:** Adrian Hillier

**To test:**
1.Open Elemental Analysis Interface
2.Put Sundials into your Data search directories in MUD (Contact myself or @AnthonyLim23)
3.Enter 2695 and click Load.
4.Plot detector GE1
5.Select minor peaks
6.Deselect minor ticks (or infact the major ticks!)

Fixes #29162 . 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.m [... cut]
closed 2020-09-15 by martyngigg