mantidproject/mantid #29165: SliceViewer: zoom on non-orthogonal view does not increase view occupancy of image

martyngigg, 2020-08-11:
Zooming on the non-orthogonal sliceviewer does not act as one would expect. On the orthogonal view as you zoom the view fills with more of the data but with the nonorthogonal view the data in the view seems to remain constant:

[image]

# Steps To Reproduce

The following script produces a workspace that allows angles to be played with:

```python
from mantid.simpleapi import *

SXD23767 = Load(Filename='SXD23767.raw', LoadMonitors='Exclude')
# Set some UB with angles we can play with
SetUB(SXD23767, 1,1,2,90,90,120)
mdws = ConvertToDiffractionMDWorkspace(InputWorkspace='SXD23767',
                                       OutputDimensions='HKL')
```

* Open in the sliceviewer
* Click non-orthogonal view
* Zoom in on a position by scrolling

RichardWaiteSTFC, 2020-08-11:
Have found that when the viewing axes are set to (x,y) = (K,L) in non-orthogonal view, zooming or selecting a peak in the table will cause an exception that also seems to be related the axes limits.
[image]

RichardWaiteSTFC, 2021-01-11:
In process of testing in issue #30035, I have discovered that if you pan the view you can bring the data back into the binned region - it's just that the extent of the plotted data is reduced with the zoom

PR #30342 (2021-01-14): Fixed bug to get correct extent of bin arguments from axes limits of sliceviewer
**Description of work.**

Fixed bug that was not getting the proper extent for binning arguments when rebinning non-orthogonal data based on axes limits in sliceviewer. The y-axis is not aligned with the vertical plot axis which wasn't reflected in the determination of the limits (xmin,xmax).

**To test:**
(1) Make a non-orthog MD workspace with this script
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-5,5,-5,5',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       [... cut]
closed 2021-01-14 by gvardany