mantidproject/mantid #29197: Muon Analysis Fitting tab not reenebaled on phase quad error

Mantid-Matthew, 2020-08-19:
When there is an error in the calculation of the phase table or phase quad calculation the fitting tab is not correctly re-enabled. An easy way to get an error in the phase table calculation is to only have one group in the grouping tab which will then mean that a phase table calculation is attempted with a missing group.

AnthonyLim23, 2020-08-19:
should we only enable it if there is more than 1 group?

Mantid-Matthew, 2020-08-19:
I think this is more of a generic issue where if there is any error in the calculation the the signal to enable disabled tabs is not sent. It actually handles things quite gracefully otherwise.

Matthew-Andrew, 2020-09-02:
I suspect this is due to the error callback not re-enabling the GUI.

closed 2021-03-01 by AnthonyLim23