mantidproject/mantid #29198: Launch Fail: MacOS Workbench Nightly

DanielMurphy22, 2020-08-19:
**Original reporter:** Chris Ridley / ISIS

Mantid Nightly Fails to open on MacOS. A local build in this interval (24th July to Now [19th Aug]) opens fine, and a local package build seems to be fine also! The downloadable nightly packages have this launch failure!

Output Stacktrace: 
```
Traceback (most recent call last):
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantidqt/widgets/jupyterconsole.py", line 21, in <module>
    from qtconsole.inprocess import QtInProcessKernelManager
  File "/Applications/MantidWorkbench.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/qtconsole/inprocess.py", line 5, in <module>
    from ipykernel.inprocess import (
ModuleNotFoundError: No module named 'ipykernel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/start.py", line 192, in start
    exit_value = start_workbench(app, options)
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/start.py", line 127, in start_workbench
    main_window.setup()
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/mainwindow.py", line 162, in setup
    from workbench.plugins.jupyterconsole import JupyterConsole
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/plugins/jupyterconsole.py", line 20, in <module>
    from mantidqt.widgets.jupyterconsole import InProcessJupyterConsole # noqa
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantidqt/widgets/jupyterconsole.py", line 24, in <module>
    from IPython.qt.inprocess import QtInProcessKernelManager
ModuleNotFoundError: No module named 'IPython.qt.inprocess'
```
### Steps to reproduce the behavior
Download a nightly for Mac and try to open Workbench.
The first version that is broken is `mantid-5.0.20200723.1804-HighSierra.dmg`
The last that works is `mantid-5.0.20200720.1658-HighSierra.dmg`
### Platforms affected
Mac Workbench

DanielMurphy22, 2020-08-19:
Launching with step 3b of troubleshooting still fails and gives:
```
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.8/3.8.3/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/Cellar/python@3.8/3.8.3/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/main.py", line 55, in <module>
    main()
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/main.py", line 46, in main
    start(options)
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/main.py", line 50, in start
    from workbench.app.start import start
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/start.py", line 15, in <module>
    from mantid.api import FrameworkManagerImpl
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantid/__init__.py", line 105, in <module>
    from mantid import kernel as _kernel
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantid/kernel/__init__.py", line 36, in <module>
    import_mantid_cext('._kernel', 'mantid.kernel', globals())
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantid/utils.py", line 35, in import_mantid_cext
    lib = import_module(modulename, package)
  File "/usr/local/Cellar/python@3.8/3.8.3/Frameworks/Python.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
SystemError: initialization of _kernel raised unreported exception
```

DanielMurphy22, 2020-08-19:
Launching MantidPlot works but shows this error:
```
The `IPython.qt` package has been deprecated since IPython 4.0. You should import from qtconsole instead.
ModuleNotFoundError: No module named 'IPython.qt.inprocess'
  at line 1 in '<Interface>'
  caused by line 9 in '/Applications/MantidPlotBroken.app/Contents/MacOS/ipython_widget/__init__.py'
  caused by line 36 in '/Applications/MantidPlotBroken.app/Contents/MacOS/ipython_widget/mantid_ipython_widget.py'
```

DanielMurphy22, 2020-09-02:
The latest nightly (mantid-5.0.20200814.1501-HighSierra.dmg) is not fixed, so reopening. Here is the new, very similar output:

```
Traceback (most recent call last):
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantidqt/widgets/jupyterconsole.py", line 21, in <module>
    from qtconsole.inprocess import QtInProcessKernelManager
  File "/Applications/MantidWorkbench.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/qtconsole/inprocess.py", line 5, in <module>
    from ipykernel.inprocess import (
ModuleNotFoundError: No module named 'ipykernel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/start.py", line 196, in start
    exit_value = start_workbench(app, options)
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/start.py", line 131, in start_workbench
    main_window.setup()
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/app/mainwindow.py", line 162, in setup
    from workbench.plugins.jupyterconsole import JupyterConsole
  File "/Applications/MantidWorkbench.app/Contents/MacOS/workbench/plugins/jupyterconsole.py", line 20, in <module>
    from mantidqt.widgets.jupyterconsole import InProcessJupyterConsole # noqa
  File "/Applications/MantidWorkbench.app/Contents/MacOS/mantidqt/widgets/jupyterconsole.py", line 24, in <module>
    from IPython.qt.inprocess import QtInProcessKernelManager
ModuleNotFoundError: No module named 'IPython.qt.inprocess'
```

martyngigg, 2020-09-04:
We've worked out that this is something to do with a single builder: `isis-mantid-mac-pro`. Leaving the issue open so it doesn't get forgotten but the code is okay as `isis-ndw1170` produces a working version.

martyngigg, 2020-09-04:
I think I have solved this. There was a version of `ipykernel` in `~/Library/Python/3.8/lib/python/site-packages` so the `pip` thought it didn't need installing. I've removed this so that it can build.

PR #29240 (2020-08-27): Fix RPATH for MantidWorkbench on macOS
**Description of work.**

Fixes startup of macOS Mantidworkbench executable.

**To test:**

I have built a package so this can be tested fully: https://builds.mantidproject.org/job/pull_requests-osx/37292/artifact/build/mantid-5.0.20200827.708-HighSierra.dmg

Install it and start both MantidPlot & MantidWorkbench

Fixes #29198 

*This does not require release notes* because **it is a regression this dev cycle**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

## [... cut]
closed 2020-09-04 by martyngigg