mantidproject/mantid #29265: Indirect DA loading hard crash

DanielMurphy22, 2020-09-02:
**Original reporter:** [Mantid Forum](https://forum.mantidproject.org/t/indirect-diffraction-interface-not-processing-files-from-osiris-or-iris/677)

Current workaround is loading the data from the archive works

### Expected behavior
The indirect data analysis interface should be able to load .nxs and .raw files safely on all platforms
### Actual behavior
On Windows:
browsing to OSIRIS00142072.nxs :white_check_mark: (reads file and runs the reduction successfully)
browsing to OSIRIS00142072.raw :x: (hard crash)
Vice versa on MacOS:
browsing to OSIRIS00142072.nxs :x:
browsing to OSIRIS00142072.raw :white_check_mark:

### Steps to reproduce the behavior
Download the files to your local computer
Open Workbench, IndirectDataAnalysis interface
Try loading by Browsing to the file

### Platforms affected
All for different file types.

SpencerHowells, 2020-09-02:
@DanielMurphy22 The interface giving the problem is Indirect>Diffraction. Indirect>DataAnalysis is not designed to read raw files but reduced files and not diffraction. It has a filter to only accept the correct workspaces.

AnthonyLim23, 2020-09-07:
I have just had a go at the data reduction GUI with the nightly and raw files. It seems to work ok.

robertapplin, 2020-09-08:
This looks similar to a problem I have found in data reduction #29346 .

AnthonyLim23, 2020-09-08:
I think it is a little different. This issue is that the GUI remains greyed out forever, because it cannot find the information

SpencerHowells, 2020-09-08:
@AnthonyLim23 @robertapplin The Mantid Help problem was about reducing Diffraction data. After some effort on my part the user found out that the culprit was a hyphen in the directory name. The issue refers to the data analysis interface not reading .raw files. The interface is NOT meant to read raw files. There are filters to prevent this being attempted - for example, the nxs file name should end with _red!

DanielMurphy22, 2020-09-14:
Vicky Garcia-Sakia has found a similar behaviour with loading raw files in Indirect Data Reduction.

I think for both interfaces, as Spencer mentions, we should restrict the user input if they only take .nxs files. It is worth checking that this behaviour is the same for MacOS!

AnthonyLim23, 2020-09-14:
I think Vicky is doing the same thing as Sanghamitra. When using raw files we also have additional files that need to be read. It is assumed that these files exist in the same directory as the raw file (this is the setup on the archive). If you place them in different place it can (but not always) result in crash.

If we do anything for this it should be catching the crash and giving a warning.

SpencerHowells, 2020-09-14:
@AnthonyLim23 @DanielMurphy22 Vicky's situation was a bit more subtle/complicated. The *Event*.nxs file was missing. For early runs this was a separate file while for newer runs it is incorporated into the main nxs file. So any error message should say that ALL files in the Archive associated with a run number must be copied to the same local directory. Good practice, which some of us adhere to, has all raw files in the same directory!

AnthonyLim23, 2020-11-24:
@Harrietbrown can we update the error message for this release?

DanielMurphy22, 2020-11-24:
I wonder has this been fixed by https://github.com/mantidproject/mantid/pull/29734 and https://github.com/mantidproject/mantid/pull/29585 ???

SpencerHowells, 2020-11-24:
@AnthonyLim23 @DanielMurphy22 #29585 should cure the main problem. Bu problems could still occur if a directory contains '.', '-' or a space (as in Windows default 'My PC'.

Harrietbrown, 2020-11-30:
Neither of those files load successfully for me in data analysis. both work in data reduction however.

AnthonyLim23, 2020-11-30:
@Harrietbrown that sounds right, it is raw data so needs reducing

AnthonyLim23, 2021-01-15:
@Harrietbrown and @SpencerHowells do we need to look into this still?

SpencerHowells, 2021-01-15:
@AnthonyLim23 @Harrietbrown This is done. Can be closed.

closed 2021-01-18 by AnthonyLim23