mantidproject/mantid #29291: Unscripted Testing Workbench Group 1

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/Core/Core.html

Matt-Cumber, 2020-09-07:
Delete Workspace Setting needs warning for a restart of workbench #29333

Matt-Cumber, 2020-09-07:
Potential bug in instrument view. Load data and open instrument view. Use mouse to rotate instrument. change from full 3d to anything else so you can no longer rotate. change back to full 3d and you still cant rotate even though you should be able to. switch tabs and go back to render and you can again

Matt-Cumber, 2020-09-08:
Docs of PlotAsymmetryByLogValue: https://docs.mantidproject.org/nightly/algorithms/PlotAsymmetryByLogValue-v1.html

Example - Calculating asymmetry for a series of MUSR runs using a range:
Does nor work as usage data provided does not include MUSR00015195

Matt-Cumber, 2020-09-08:
Slice viewer X and Y buttons are weird

Matt-Cumber, 2020-09-08:
Potentially start up window does not appear on workbench. need to check this over

Matt-Cumber, 2020-09-09:
Unhandled exception plotting spectrum with errors then normalise to bin width

Matt-Cumber, 2020-09-14:
Closing now as issues above have been addressed

closed 2020-09-14 by Matt-Cumber