mantidproject/mantid #29294: Unscripted Testing Workbench Group 4

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/Core/Core.html

joseph-torsney, 2020-09-07:
Plots with `axvline` or `axhline` do not persist between project save and load.
script
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
import matplotlib.pyplot as plt
from mantid.api import AnalysisDataService as ADS

ws1 = CreateSampleWorkspace(OutputWorkspace="ws1")
fig, axes = plt.subplots(edgecolor='#ffffff', num='SANSLOQCan2D-1', subplot_kw={'projection': 'mantid'})
axes.plot(ws1, color='#ff7f0e', label='ws1', specNum=2)
axes.axhline(0.1, xmin=0, xmax=1)
plt.show()
```

error `A plot was unable to be loaded from the save file. Error: list `index out of range`

joseph-torsney, 2020-09-07:
plot axes and labels get 'stuck' when zooming in and out, so it looks like there's two of them 
[image]

joseph-torsney, 2020-09-07:
slice viewer - switching axes doesn't change the plot until you switch it back, and then the axis ranges are the wrong way around

joseph-torsney, 2020-09-07:
slice viewer - docs say there is a 'view' menu, there aren't any menus on the top bar for me (windows)

joseph-torsney, 2020-09-07:
changing color of curve on a plot, then saving project. lots of funny lines on load 
[image]

closed 2020-09-14 by joseph-torsney