mantidproject/mantid #29297: Unscripted Testing Muon

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2020-09-10:
ALC:

It doesn't seem to update when changing from a single string of runs (1-20) to leaving some out (1-9,16-20)

On base line modelling and a function and region. Do the fit, change the range do another fit. Then go to the corrected data tab. There are multiple instances of the same data point. We should only display the latest one

AnthonyLim23, 2020-09-10:
Elemental analysis:

Load data, plot it, tick Gamma, click Fe. Untick Fe- crash (probably related to https://github.com/mantidproject/mantid/issues/29162

AnthonyLim23, 2020-09-10:
FDA + MA:

In results table the output no longer has the y errors set in the table columns 

External plots do not include the grid lines (I dont care)

In TF asymmetry mode we can activate the `select data to fit` button by turning TF asymmetry on and off. We should make sure it remains disabled (using it can crash Mantid)

AnthonyLim23, 2020-09-10:
MA and FDA doesnt reset properly after changing instrument or pressing clear

closed 2020-09-14 by AnthonyLim23