mantidproject/mantid #29304: Unscripted Testing FilterEvents

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

martyngigg, 2020-09-17:
@RichardWaiteSTFC Do you think you are done with the unscripted testing here? If so, could we close this issue?

RichardWaiteSTFC, 2020-09-17:
Created low priority issues #29475 and #29473 - behaviour hasn't changed in last couple release.

closed 2020-09-17 by RichardWaiteSTFC