mantidproject/mantid #29312: Unscripted Testing ISIS Reflectometry

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

For testing procedures see http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

DannyHindson, 2020-09-09:
Two issues found so far:
- the shortcut on the applications menu in Ubuntu isn't created. This is because the file installers\LinuxInstaller\launcher-template.desktop.in has had a C++ header (the copyright statement) added and Ubuntu doesn't seem to like this. If I delete the comment the menu item shows up
- the Reflectometry UI crashes (and Mantid Workbench as a whole crashes) after loading a .json file from the Batch, Load menu. I've tested this on WIndows and Ubuntu and it happens on both. After loading the .json file I see the crash happening 20-30 secs later:
[image]

In addition Workbench crashes on shutdown for me even if you've not visited the reflectometry UI on Windows giving this stack trace:
[image]

gemmaguest, 2020-09-09:
Yes, this crash is a genuine problem. It's actually because the sample data is slightly out of date and contains only 9 columns on the experiment settings table when there are now 10. It should be backwards compatible with old files, though, so this should work. The crash is happening in the encoder which is why it's taking a while - it's happening when project recovery takes a snapshot.

DannyHindson, 2020-09-11:
Just found another issue. The reflectometry user interface crashes when I'm doing the following steps:
- search for OFFSPEC 1110285
- transfer run 10797 over the right hand side
- cut the 10797 row
- try to transfer 10797 again
Problem comes from unhandled exception raised in RowLocationAdapter::indexAt

DannyHindson, 2020-09-11:
When running through the "Closing the interface" part of the unscripted test I didn't get a Save\Discard\Cancel dialogue:
- load INTER_NR_test.json
- process some of the groups
- delete a row from a group
- edit some of the cells in a row

DannyHindson, 2020-09-11:
When testing the "Export as ASCII" part of the script I found that when you select a small number of parameters from the "List Of Logged Parameters" you get some rows with content "Parameter : Not Defined" output into the text file:
[image]

Seems to be when you select fewer than 9 parameters. Not sure if this is expected but seems a bit odd - especially the <9 part of the behaviour.

gemmaguest, 2020-09-14:
> When testing the "Export as ASCII" part of the script I found that when you select a small number of parameters from the "List Of Logged Parameters" you get some rows with content "Parameter : Not Defined" output into the text file:

I agree this is a little odd but I think it is fine. This is an expected change because we have changed to use the `SaveReflectometryAscii` algorithm. I believe the 9 parameters are expected to always be included as part of the ILL(MFT) format even if empty - this was validated at ILL so I assume it is fine, and our scientists have tried this out and not complained! We agreed that we would just use the same header for the Custom format. The old output formats had numerous errors so were not really usable, so either way this is still an improvement so not something I am worried about for this release.

DannyHindson, 2020-09-16:
Closing the issue now that unscripted testing complete. Separate issues created for the problems found

closed 2020-09-16 by DannyHindson