mantidproject/mantid #29315: Unscripted Testing  Mantid Basics Course

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Build a docs-html target of Mantid
* Check against the online documentation (http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/index.html#mantid-basic-course)
* Open up the basic course (docs/html/tutorials/mantid_basic_course/index.html)
* Check that the pages in there make sense

DanielMurphy22, 2020-09-07:
This docs edit looks like its in the wrong place: https://github.com/mantidproject/mantid/pull/28771/commits/0389b7b3bbb56ea02c98d0150fed31dfca59e3f3

sf1919, 2020-09-07:
Just reviewing the Workbench Main Window screen shot in comparison to current version. I don't feel there is any point updating the screen shot from Version 4.2.0 as nothing has changed apart from the version number. 

However in the screen shot the message reads
`Welcome to Mantid 4.2.0
Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid4.2`

Whereas the testing version doesn't seem to ask you to refer to a specific version of Mantid
`Welcome to Mantid 5.0.20200905.1517
Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
All instrument definitions up to date`

Do we need this message to ask users to refer to version 5.0 or is this no longer necessary?

sf1919, 2020-09-07:
In Loading Data under Types of Data Files the wording is slightly confusing. It starts talking about data types and then states
`Fortunately, you don’t have to learn how to use all of these Load algorithms.`
It's the first time Load algorithms are mentioned and could potentially be confusing to users with less computing/algorithm experience. Perhaps we need a sentence before this explain that each different data format requires a different load algorithm. Then the sentence about not needing to know how to use all the Load algorithms makes a bit more sense.

Its a minor thing so can be bumped to version 6.

sf1919, 2020-09-07:
Another one in Loading Data (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/01_loading_data.html#loading-data).

This time it's the section on `Loading Lots of Data Files`. It talks about using the Load command which isn't covered previously. This is a bit confusing and possibly needs rewording to make it clearer. Its possible to do these commands in the box for 'File' in the Load Dialog window, rather than browsing for files. Is this what is meant rather than using the script editor?

sf1919, 2020-09-08:
In the Viewing Data part of the course (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/02_the_matrix_workspace.html) specifically `Workspace Matrix View - Show Data` it says that after clicking `Show Data` the dialog gives you options and just to press OK. No dialog appears. Should a dialog appear? Or does it no longer exist and the instructions need updating?

sf1919, 2020-09-08:
In Displaying 1D Data (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html
) the initial instructions take you through how to plot spectra 1-3. Can we include the step to click OK in this numbered list? Just to make it clear not to click Plot All? Relatively minor point and if you click Plot All you do get a warning about plotting all 922 spectra so hopefully that would stop people going down this route.

sf1919, 2020-09-08:
Another one in Displaying 1D Data - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html

In the section 'Another way to plot' step 4 is `Please highlight rows 0-2 and bring up the right-click menu:` . After this is an image which shows which value to select. Can we also have it written out which of the menu items to select? i.e. Plot spectrum (values only). This isn't accessible to people using screen readers. An alternative would be to put the instruction in the alt text of the image. Currently alt is `../../../_images/300px-MatrixDisplayRightClickOptions.png` which is not an alternative description at all.

sf1919, 2020-09-08:
One of the screenshots on the Formatting Plots with the User Interface page is out of date - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html
.

Under Options Menu for 1D plots the screenshot of the Axes tab doesn't include Show Minor Ticks, Show Minor Gridlines or the Apply to all button.

sf1919, 2020-09-08:
Following on from the previous comment there is another screenshot that may need to be changed on Basic 1D and Tiled plots page (https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#basic-1d-plots). 

Under File > Settings at the bottom of the page the screenshot of the plots dialog doesn't include Show Minor Ticks and Show Minor Gridlines. It's also missing the sections Default Font, Legend and Images.

sf1919, 2020-09-09:
Displaying and Navigating to the Instrument View - https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/02_displaying_and_navigating.html

Under `Opening the Instrument View` the second direction isn't clear. Right clicking on the Workspace group (`MUSR00015189`) will not give you the `Show Instrument` option. Needs to be made clearer that you need to click one of the specific workspaces in this group e.g. `MUSR0001589_1`.

DanielMurphy22, 2020-09-09:
In instrument viewer, The bottom bar which shows the X axis (unit usually time) for the workspace affects this. It seems that:
- when this bottom bar (displaying the range of time over which the detector display is integrated) is the full range, then your two points are true (Apply and Save is not greyed out, and the top button is  Apply detector mask to View )
- when this bar is less than the full possible range (either drag one side inwards OR type an new value in the start or end value boxes), the Apply and Save button is greyed out and the top button is Apply bin mask to View
Maybe this should be clarified briefly on [this page](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/04_masking_and_grouping.html)!

sf1919, 2020-09-10:
In the fitting exercises (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/05_fitting_exercises.html) Exercise 3, there seems to be a step missing. We need to set a to 0.409108 otherwise the graph turns out nothing like the image! Also the text says to Tie but I found that to get the result in the screen shot you need to select Fix from the menu, not Tie.

sf1919, 2020-09-10:
Similar to the comment above about Minor Ticks and Minor Gridline options not on the screenshots for formatting plots they are also not on the screenshot for Mantid Settings - Plots. See https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#save-settings

martyngigg, 2020-09-17:
@sf1919  Do you think you are done with the unscripted testing here? If so, could we close this issue?

sf1919, 2020-09-17:
@martyngigg I finished the unscripted testing of the Basic Course last week. I did find an issue in the other training I'm doing at the moment which @DanielMurphy22 has linked to this one but I'm not working through the material as an unscripted test. So I think we can close this but can @DanielMurphy22 give a 2nd opinion?

DanielMurphy22, 2020-09-17:
Yes this is fine to close, and I have mentioned these comments in another issue: #29455

closed 2020-09-17 by DanielMurphy22