mantidproject/mantid #29316: Unscripted Testing M-slice

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start, this should probably be tested in iDAaaS

DanielMurphy22, 2020-09-07:
Load data and Plot slice (I'm using LET36360_1.05meV_rings.nxspe).
Change the colorbar scale to Log and then back to Lin and the original and final plots are different!
Is this just related to this section where its mentioned that some pixels may be left white??
 https://mantidproject.github.io/mslice/quickstart.html#plotting-a-slice

[image]
[image]

martyngigg, 2020-09-09:
It's not immediately clear to me why that happens but I would expect it to look the same on the flip back. The second picture looks like it has way more NaNs in it...

DanielMurphy22, 2020-09-09:
Tested on MacOS and IDAaaS, no further issues found.

closed 2020-09-09 by DanielMurphy22