mantidproject/mantid #29318: Unscripted testing Error Reporter

DanielMurphy22, 2020-09-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

Matt-Cumber, 2020-09-07:
MantidPlot error reporter not showing after test 1: #29329

Matt-Cumber, 2020-09-08:
If you enter text in additional information box "Share non-identifiable information" button greyed out, deleting text does not un-grey out button. 

If you delete all text then enter one character the button is clickable again.

If you a character in Name or Email then delete it button is clickable again

closed 2020-09-10 by Matt-Cumber