mantidproject/mantid #29324: Muon Easier Tab Undocking

DanielMurphy22, 2020-09-04:
**Original reporter:** [Mantid Forum](https://forum.mantidproject.org/t/muon-interface-tabs-keep-leaving-parent-window/681)

It would be nice for unintentionally undocking the tabs in the Muon Analysis interface to be less likely. Currently, clicking to change tabs can undock the new tab, especially when using a trackpad! 

Disallowing click + drag undocking and forcing undocking to only work by double-clicking would solve this!

### Steps to reproduce the behavior
Open Muon Analysis interface and notice that you can undock the tabs (e.g. Home, Grouping...) by either:
- Click and drag
- Double-clicking

Using a laptop trackpad can mean simply clicking (and not intentionally dragging) can undock the tab.

### Platforms affected
Noticed on MacOS and IDaaAS, but probably all!

AnthonyLim23, 2020-09-07:
I think this is a duplicate issue.

AnthonyLim23, 2020-09-07:
https://github.com/mantidproject/mantid/issues/27052
https://github.com/mantidproject/mantid/issues/26613

I know we have looked at this briefly and it was not trivial to fix what we currently have. We did have an old version that needed the user to click a box to undock. 

Some users really like the undocking, as it makes it look like Wimda.

robertapplin, 2021-01-29:
This came up in beta testing with James.

robertapplin, 2021-07-28:
This was fixed as part of #31455

closed 2021-07-28 by robertapplin