mantidproject/mantid #29325: Context menu for plotting 1D MDHistoWorkspaces in workbench

AndreiSavici, 2020-09-04:
1D plots from an MDHistoWorkspace are not available. Or from MDHistoWorkspace with only one non-integrated dimension.

### Expected behavior
Plot 1D MD workspace from workbench (we can do it from a script)

### Actual behavior
Only slice viewer is available as a context menu, but that complains about not having at least 2 non-integrated dimensions

### Steps to reproduce the behavior
Create a 1D MDH workspace in workbech.
```python
test = CreateMDHistoWorkspace(SignalInput='1,2,3,4,2,1',
                              ErrorInput='1,1,1,1,1,1',
                              Dimensionality=1,
                              Extents='0.5,6.5',
                              NumberOfBins=6,
                              Names='|Q|',
                              Units='AA^-1')
```
Try to right and plot it
### Platforms affected
All

Harrietbrown, 2020-09-10:
On further inspection this issue is more complecated than simply exposing the workpsace to the plot option in the context menu. the amount of work required to get this sorted is more than should be done in a release sprint.

PlotMD was in mantidplot but obviously is not available any more.

wdzhou, 2020-09-20:
Some observation for the current implementation.

1. If two workspaces of different types are selected, the pop up menu is determined by the type of the second workspace in the order of tree.
1.

closed 2020-09-25 by martyngigg